realsenseai/librealsense #184: error message

ge53, 2016-06-26:
| Required Info  |   |
|---|---|
| Camera Model | R200, F200 or SR300 | 
| Firmware Version |   | 
| Operating System & Version |   |
| Kernel Version (Linux Only)  |   |
| Build System |  VS2013, XCode or Makefile | 

SR 300, Mac ElCapitan10.11.50, XCode
When trying to compile XCode examples I get the folowing error:
_dyld: Library not loaded: /usr/local/lib/libusb-1.0.0.dylib
  Referenced from: /Users/geraldineerman/Library/Developer/Xcode/DerivedData/c-tutorial-1-depth-dmbdcchtfexvyaesbopvojjjccyo/Build/Products/Debug/c-tutorial-1-depth
  Reason: Incompatible library version: c-tutorial-1-depth requires version 2.0.0 or later, but libusb-1.0.0.dylib provides version 1.0.0
(lldb)_ 
I did follow your OSX directions re homebrew and lib installs...any help/suggestions,please? 
Thanks

ddiakopoulos, 2016-06-26:
This doesn't appear to be an easily-replicable issue. I would start by Googling around `Reason: Incompatible library version` or `Reason: Incompatible library version libusb` From a very quick survey of the results, it seems there's something funky going on with brew and possibly an older install of libusb on your system.

ge53, 2016-06-26:
thanksfor your quick response---reinstalled libusb , getting new error : 
_Ld /Users/geraldineerman/Library/Developer/Xcode/DerivedData/librealsense-dpitndvxkguxwncwadjitpicdtbn/Build/Products/Debug/librealsense.dylib normal x86_64
    cd /Users/geraldineerman/Desktop/librealsense-master/librealsense.xc
    export MACOSX_DEPLOYMENT_TARGET=10.10
    /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/clang++ -arch x86_64 -dynamiclib -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.11.sdk -L/Users/geraldineerman/Library/Developer/Xcode/DerivedData/librealsense-dpitndvxkguxwncwadjitpicdtbn/Build/Products/Debug -L/usr/local/lib -F/Users/geraldineerman/Library/Developer/Xcode/DerivedData/librealsense-dpitndvxkguxwncwadjitpicdtbn/Build/Products/Debug -filelist /Users/geraldineerman/Library/Developer/Xcode/DerivedData/librealsense-dpitndvxkguxwncwadjitpicdtbn/Build/Intermediates/librealsense.build/Debug/librealsense.build/Objects-normal/x86_64/librealsense.LinkFileList -mmacosx-version-min=10.10 -Xlinker -no_deduplicate -stdlib=libc++ -lusb-1.0.0 -Xlinker -dependency_info -Xlinker /Users/geraldineerman/Library/Developer/Xcode/DerivedData/librealsense-dpitndvxkguxwncwadjitpicdtbn/Build/Intermediates/librealsense.build/Debug/librealsense.build/Objects-normal/x86_64/librealsense_dependency_info.dat -o /Users/geraldineerman/Library/Developer/Xcode/DerivedData/librealsense-dpitndvxkguxwncwadjitpicdtbn/Build/Products/Debug/librealsense.dylib_

ld: file too small (length=0) file '/usr/local/lib/libusb-1.0.0.dylib' for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)

ge53, 2016-06-26:
thanksfor your quick response---reinstalled libusb , getting new error :

*Ld
/Users/geraldineerman/Library/Developer/Xcode/DerivedData/librealsense-dpitndvxkguxwncwadjitpicdtbn/Build/Products/Debug/librealsense.dylib
normal x86_64cd
/Users/geraldineerman/Desktop/librealsense-master/librealsense.xcexport
MACOSX_DEPLOYMENT_TARGET=10.10/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/clang++
-arch x86_64 -dynamiclib -isysroot
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.11.sdk
-L/Users/geraldineerman/Library/Developer/Xcode/DerivedData/librealsense-dpitndvxkguxwncwadjitpicdtbn/Build/Products/Debug
-L/usr/local/lib
-F/Users/geraldineerman/Library/Developer/Xcode/DerivedData/librealsense-dpitndvxkguxwncwadjitpicdtbn/Build/Products/Debug
-filelist
/Users/geraldineerman/Library/Developer/Xcode/DerivedData/librealsense-dpitndvxkguxwncwadjitpicdtbn/Build/Intermediates/librealsense.build/Debug/librealsense.build/Objects-normal/x86_64/librealsense.LinkFileList
-mmacosx-version-min=10.10 -Xlinker -no_deduplicate -stdlib=libc++
-lusb-1.0.0 -Xlinker -dependency_info -Xlinker
/Users/geraldineerman/Library/Developer/Xcode/DerivedData/librealsense-dpitndvxkguxwncwadjitpicdtbn/Build/Intermediates/librealsense.build/Debug/librealsense.build/Objects-normal/x86_64/librealsense_dependency_info.dat
-o
/Users/geraldineerman/Library/Developer/Xcode/DerivedData/librealsense-dpitndvxkguxwncwadjitpicdtbn/Build/Products/Debug/librealsense.dylib*

ld: file too small (length=0) file '/usr/local/lib/libusb-1.0.0.dylib' for
architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see
invocation)

http://myvernacular.tumblr.com

ge53, 2016-06-27:
Reinstalled libusb yet again and fixed linkages with "brew doctor" /got Xcode examples working. Thanks for your help!

ddiakopoulos, 2016-06-27:
Great! Glad you fixed it :)

closed 2016-06-27 by ddiakopoulos

dimitre, 2019-01-08:
I can't run either even in 10.13.6, XCode 9.4.1
ld: library not found for -lusb-1.0