realsenseai/librealsense #185: How can I get DisparityConstant in librealsense

wayy12, 2016-06-27:
I'm porting my application based in RSSDK to ubuntu use librealsense.
But I can't get the constant "DisparityConstant".In RSSDK,It get "with __inline pxcF32 QueryDSZToDisparityConstant(void)".
So,what is the similar API in librealsense?
Or,Can I get baseline value?

TKS!

wayy12, 2016-06-27:
@brief When in a rectified mode, ZToDisparityConstant = baseline * focalLengthX, 
		where baseline is measured in mm, and focalLengthX is measured in pixels.
		Used to convert between Z distance (in mm) and disparity (in pixels), where ZToDisparityConstant / Z = disparity.
        @return the z to disparity constant
        */
        __inline pxcF32 QueryDSZToDisparityConstant(void) {
            pxcF32 value = PROPERTY_VALUE_INVALID;
            QueryProperty(PROPERTY_DS_Z_TO_DISPARITY_CONSTANT, &value);
            return value;
        }

ddiakopoulos, 2016-06-27:
If you enable Points stream (`xyz32f` format) then this is already taken care of in the deprojection.

Otherwise, it's still quite easy to get the baseline. Ask the API for the extrinsics between the infrared1 and infrared2 stream. The baseline will be in the X translation component.

closed 2016-06-30 by ddiakopoulos