realsenseai/librealsense #190: installation.md: Kernel patch still necessary with 4.4.0-28?

mintar, 2016-06-29:
| Required Info  |   |
|---|---|
| Camera Model | R200 | 
| Firmware Version | 1.0.72.06  | 
| Operating System & Version | Ubuntu 14.04  |
| Kernel Version (Linux Only)  | 4.4.0-28-generic  |
| Build System |  Makefile | 

I have just updated to the latest stock 4.4 kernel (via `sudo apt-get install linux-image-generic-lts-xenial` on Ubuntu 14.04) and tested the RealSense R200 with the ROS driver. Everything works, as far as I can tell. So my question is: is the patch still necessary? How would I notice if I'm missing the patch? The kernel sources for 4.4.0-28-generic show that `realsense-camera-formats.patch` has already been partially applied.

In 9a00ba5, @ddiakopoulos has added information on how to patch the kernel for Ubuntu 16.04 (which should be the same kernel I installed), so that's what got me confused. Otherwise I would have opened a pull request for the documentation.

MichaelKorn, 2016-06-29:
That is interesting. The Patch defines new color spaces. Without this patch you can grab images from the camera, but the color image and/or depth image look like datatrash (I don't remember which image it was.).

mintar, 2016-06-29:
As I said, I can confirm that the ROS driver works on that kernel. I get correct colored point clouds, depth and color images. It looks like the important parts of the patch have already been applied upstream. If you do this (on Ubuntu 14.04):

    apt-get source linux-image-4.4.0-28-generic
    patch -p1 < scripts/realsense-camera-formats.patch

... you can see that one of the 3 files in the patch is not touched at all, which I think means that this part of the patch is already upstream.

MichaelKorn, 2016-06-30:
I updated the kernel with Ubuntu 14.04 to 4.4.0-28 with:
`sudo apt-get install linux-generic-lts-xenial linux-headers-generic-lts-xenial linux-image-generic-lts-xenial linux-tools-generic-lts-xenial`
I also compared the source code with the patch. Some color spaces are part of the kernel , now (e.g. Z16). Nevertheless depth image and IR image are broken with SR300, I still need to apply the patch.
The camera is running much more stable. With Kernel 3.16 the hole machine hangs regularly when using the SR300.

ddiakopoulos, 2016-06-30:
`4.4.0-28` appears to the the first kernel version available in apt-get with the additional R200 formats. Let me be very clear: **`4.4.0-28` only has the format patch for the R200 camera only**. I knew this kernel would be available soon, but there was a string of 5-6 issues posted on this Github re: confusion whether to apply the patch, and one of the contributors of librealsense prematurely noted in the installation guide that this kernel was already shipping as part of the 16.04 installer (it was not, but should be in the next ~10 days). There's an equal number of R200 and SR300 users of librealsense, and by far most issues reported on Github are purely installation related -- my goal is to keep it as simple as possible.

Since the installation guide is long enough already, my recommendation in `installation.md` is that **everyone** apply the patch just to be sure. Applying the patch on top of the `4.4.0-28` kernel will not negatively impact the system and will still apply the SR300-specific formats (which will likely never be upstreamed).

Hope that answers your question definitively.

closed 2016-06-30 by ddiakopoulos

mintar, 2016-07-01:
Thanks for the explanation @ddiakopoulos , that makes sense.

It's a pity that the SR300-specific patch will never be upstreamed. I try to avoid kernel patches wherever possible because of the burden of maintaining my own kernel with a set of patches for various hardware.

I agree that the installation guide is already long (and complicated) enough. However, I think it would make sense to add a note that the patch is not required for the R200 (at least when that's also true for the 16.04 kernel).

Installation on 14.04 is much less painful than the installation guide makes it look (for the R200). You don't even need glfw3 or qt5, if all you're interested in is running the ROS driver.

RealSenseCustomerSupport, 2019-03-06:
----------------------------------------------

@ghost F200 is EOL for long time. Please change to the latest supported D400 cameras. Thanks!

RealSenseCustomerSupport, 2019-03-07:
----------------------------------------------

[Please Ignore - RealSense system comment]