realsenseai/librealsense #194: Sample program crashed

jb892, 2016-07-05:
| Required Info  |   |
|---|---|
| Camera Model | R200 | 
| Firmware Version | 1.0.72.06 | 
| Operating System & Version | Ubuntu 16.04 LTS  |
| Kernel Version (Linux Only)  |  4.4.0-040400-generic  |
| Build System |  Qt creator  | 

After I installed the API on my PC, I tried to run the "cpp-capture" example program in Qt. However, I got an error says it can not find the librealsense.a file. Thus, I deleted the last line: "PRE_TARGETDEPS += $$DESTDIR/librealsense.a" from the .pro file and it makes program able to run. However, the program can only last for up to 3 seconds and then it crashes with these errors:
<--------------------------------------------------------------------->
Capturing DEPTH at 480 x 360, fov = 55.1 x 42.8, distortion = NONE
Capturing COLOR at 640 x 480, fov = 54.0 x 41.4, distortion = MODIFIED_BROWN_CONRADY
Capturing INFRARED at 480 x 360, fov = 55.1 x 42.8, distortion = NONE
Capturing INFRARED2 at 480 x 360, fov = 55.1 x 42.8, distortion = NONE
rs.warn: Subdevice 0 frame status 0x1
rs.warn: Subdevice 1 frame status 0x1
rs.warn: Subdevice 0 frame status 0x1
rs.warn: Subdevice 1 frame status 0x1
rs.warn: Subdevice 0 frame status 0x1
rs.warn: Subdevice 1 frame status 0x1
rs.warn: Subdevice 0 frame status 0x1
rs.warn: Subdevice 1 frame status 0x1
rs.warn: Subdevice 1 frame status 0x1
rs.warn: Subdevice 0 frame status 0x1
rs.warn: Subdevice 0 frame status 0x1
rs.warn: Subdevice 1 frame status 0x1
rs.warn: Subdevice 1 frame status 0x1
rs.warn: Subdevice 0 frame status 0x1
rs.warn: Subdevice 1 frame status 0x1
rs.warn: Subdevice 0 frame status 0x1
rs.warn: Subdevice 1 frame status 0x1
rs.warn: Subdevice 0 frame status 0x1
rs.warn: Subdevice 1 frame status 0x1
rs.warn: Subdevice 0 frame status 0x1
rs.warn: Subdevice 1 frame status 0x1
rs.warn: Subdevice 0 frame status 0x1
rs.warn: Subdevice 1 frame status 0x1
rs.warn: Subdevice 0 frame status 0x1
rs.warn: Subdevice 1 frame status 0x1
rs.warn: Subdevice 0 frame status 0x1
RealSense error calling rs_wait_for_frames(device:0xf9f8a0):
    Timeout waiting for frames.
Press <RETURN> to close this window...
<--------------------------------------------------------------------->
Can anyone help me to solve this issue? Thanks in advance!

eugenebartosh, 2016-11-05:
I have the same issue - Ubuntu 16.04 LTS, kernel module was patched.

jlgarcia75, 2017-10-03:
Hello,

If the issue is software related, we are no longer updating this software as we have moved to our latest generation, the D400 series. If you are in your warranty, a replacement can be provided.

Jesus G.
Intel Customer Support

closed 2017-10-03 by jlgarcia75