realsenseai/librealsense #195: R200 Realsense not detected on Linux Ubuntu

gabrielarpino, 2016-07-06:
Hi,

So I am trying to use my R200 Intel Realsense on my Clearpath Jackal Robot. I have tested the Realsense previously with ROS on my Linux PC and it worked. I then move the realsense to the robot (which runs Linux with the specs described below), and, after installation, I receive the following message when trying to run `./cpp_multicam`:

`No device detected. Is it plugged in?`

Also, the following message when trying to run any of the tutorials:

`There are 0 connected RealSense devices.`

These suggest that my device might not be connected, but I confirm that it is connected by running "lsusb -v -d 8086:0a80", and receiving the following output (first few lines):

```
Bus 004 Device 005: ID 8086:0a80 Intel Corp. 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               3.00
  bDeviceClass          239 Miscellaneous Device
  bDeviceSubClass         2 ?
  bDeviceProtocol         1 Interface Association
  bMaxPacketSize0         9
  idVendor           0x8086 Intel Corp.
  idProduct          0x0a80 
  bcdDevice           14.86
  iManufacturer           1 Intel Corp
  iProduct                2 Intel RealSense 3D Camera R200
  iSerial                 3 SN_2441014965
  .....

```
Then, I try to disconnect and reconnect my R200 from my clearpath Jackal  (again confirming that the USB is a 3.0 port), and look at dmesg:

```
input: Intel RealSense 3D Camera R200 as /devices/pci0000:00/0000:00:14.0/usb4/4-2/4-2:1.4/input/input25
 [ 2742.846008] usb 4-2: USB disconnect, device number 4
[ 2760.314974] usb 4-2: new SuperSpeed USB device number 5 using xhci_hcd
[ 2760.333668] usb 4-2: New USB device found, idVendor=8086, idProduct=0a80
[ 2760.333671] usb 4-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 2760.333673] usb 4-2: Product: Intel RealSense 3D Camera R200
[ 2760.333674] usb 4-2: Manufacturer: Intel Corp
[ 2760.333675] usb 4-2: SerialNumber: SN_2441014965
[ 2760.336008] uvcvideo: Found UVC 1.10 device Intel RealSense 3D Camera R200 (8086:0a80)
[ 2760.339419] uvcvideo: Found UVC 1.10 device Intel RealSense 3D Camera R200 (8086:0a80)
[ 2760.340835] uvcvideo: Unable to create debugfs 4-5 directory.
[ 2760.340903] input: Intel RealSense 3D Camera R200 as /devices/pci0000:00/0000:00:14.0/usb4/4-2/4-2:1.2/input/input26
```

This suggests that the Realsense is both connected AND being found by UVC, so I do not understand what the problem is. When I run dmesg after trying to run ./cpp-capture, this shows up:

```
[3230.551048] uvcvideo: Failed to set UVC commit control : -32 (exp. 34).
[ 3230.551086] uvcvideo 4-2:1.3: resume error -5
```

Maybe it's a clue? Let me know if anyone has any suggestions. Thank you very much for helping.

Best,

Gabriel

| Required Info  |   |
|---|---|
| Camera Model | R200 | 
| Firmware Version |  1.0.72.06 | 
| Operating System & Version |  Ubuntu 14.04.4 LTS |
| Kernel Version (Linux Only)  |  3.16.0-62-generic |
| Build System |  Makefile |

gabrielarpino, 2016-07-06:
I ran the following commands, thinking that It might have been due to sudo permissions on the usb (I didn't know which video the Realsense corresponded to so I just did all three):

```
sudo chmod 666 video0
sudo chmod 666 video1
sudo chmod 666 video2
```

and then I tried to run ./cpp-capture, but i get the following:

```

RealSense error calling rs_create_context(api_version:4):
    UVCIOC_CTRL_QUERY:UVC_SET_CUR error 2, No such file or directory
```

How might I fix this? Thanks :)

gs0622, 2016-07-06:
The problem is buggy Linux UVC driver against R200.
Read doc/installation.md is recommended to evaluate 4.4 kernel rather than 3.16; either follow the script of install_dependencies-4.4.sh, or upgrade your 14.04 applying Ubuntu-Xenial kernel (4.4 based) are quick options.

gabrielarpino, 2016-07-07:
Ok i will try that. Thanks :)

gabrielarpino, 2016-07-07:
Hi, I just reinstalled everything while following the script of install_dependencies-4.4.sh, and it still says 

`No device detected. Is it plugged in?`

With the same dmesg output as described above. Let me know if there might be any solution to this. Thanks.

gabrielarpino, 2016-07-10:
Anyone know how I could fix this?

when I run ./scripts/patch-uvcvideo-ubuntu-mainline.sh I get the following message:

```
Picking 'linux' as source package instead of 'linux-image-4.4.0-040400-generic'
NOTICE: 'linux' packaging is maintained in the 'Git' version control system at:
http://kernel.ubuntu.com/git-repos/ubuntu/ubuntu-trusty.git
```

Maybe this is what is causing my ./cpp-capture to return "No device detected". Anyone know how to solve this? Thanks.

ddiakopoulos, 2016-07-31:
@gabrielarpino How about re-applying the udev rules as specified `installation.md`? Looks permissions related since from your dmesg output the kernel has been correctly patched.

ddiakopoulos, 2016-07-31:
@gabrielarpino actually based on #208 it looks like you got it to work already. Can you document the root cause?

closed 2016-07-31 by ddiakopoulos

gabrielarpino, 2016-07-31:
Hi,

Yes, apologies. I did get it to work, there were permission issues with my
machine. I simply ran:

chmod 666 /dev/media0
chmod 666 /dev/video0

and it worked. Thanks !
ᐧ

*Gabriel Arpino*
*Robotics for Space Exploration - UofT*

*Director of Sponsorships(647) 967 6017
<%28647%29%20967%206017>arpino.gabriel@gmail.com
<gabriel.arpino@mail.utoronto.ca>*

jakeware, 2016-09-19:
This was helpful, but I had to sudo chmod 666 all of the following devices:
/dev/media0
/dev/media1
/dev/media2
/dev/video0
/dev/video1
/dev/video2

ravenblackx, 2016-09-27:
I had this same problem (I had to chmod /dev/video0 and /dev/video1, but not /dev/media*. So part way between gabriel's and jake's); it also appears that this chmod is revoked upon reboot or camera disconnect+reconnect.

I guess the more appropriate solution, therefore, is to add the user you're running the binary as to the 'video' group.

RoboEvangelist, 2017-03-20:
The issue with the permissions for /dev/ can be permanently solved by adding yourself to the dialout group. You will have to logout and then log back in before the group change is recognized.

You can do this with `sudo usermod -a -G dialout terrik` if terrik is your username.

whatnick, 2017-05-09:
$USER in the usermod command above to make it more generic

RoboEvangelist, 2017-05-09:
That's right!

mohamedosama70070, 2019-02-12:
@gabrielarpino can you guide me how to test the camera on Ubuntu ROS like you did i have the same version that have R200 and i can't make it work ???