realsenseai/librealsense #201: Frames out of sync R200

prosa100, 2016-07-07:
Using depth+color at 60hz, I get the frames to be consistent, but the color frame is delayed by 1 frame.
Running at 30hz, it is very inconsistent. Using a non-powered USB hub fixes this issue for 60hz, but not 30hz.

I am using a Macbook Pro 2015.

data generated with this code
```C++
while(true){
dev.wait_for_frames();
show_frames();
auto c_time = dev.get_frame_timestamp(rs::stream::color);
auto d_time = dev.get_frame_timestamp(rs::stream::depth);
std::cout	<< c_time-last_c_time << "\t" << d_time - last_d_time << "\t" << c_time - d_time << std::endl;
last_c_time = c_time;
last_d_time = d_time;
}
```

This is what unreliable timestamps look like (30hz, usb hub)

| color delta time | depth delta time | color delay from depth |
|---|---|---|
| 34 | 0 | 34 |
| 33 | 67 | 0 |
| 33 | 0 | 33 |
| 34 | 67 | 0 |
| 33 | 33 | 0 |
| 33 | 33 | 0 |
| 34 | 0 | 34 |
| 33 | 67 | 0 |
| 33 | 33 | 0 |
| 34 | 0 | 34 |
| 33 | 67 | 0 |
| 33 | 33 | 0 |
| 34 | 0 | 34 |

| Required Info  |   |
|---|---|
| Camera Model | R200 | 
| Firmware Version | 1.0.72.06  | 
| Operating System & Version |  Ubuntu 16.04 LTS |
| Kernel Version (Linux Only)  | 4.4.0-24-generic  |
| Build System |  Makefile |

leonidk, 2016-07-12:
This may just be a bug in the timestamp generation. The cameras are running at whatever frame rate you request and there's no frame-sized buffers to hold things. If the depth cameras timestamps are wrong (e.g the 0 delay frames are actually 33ms delays), everything would actually be completely synchronized. 

R200 pulls timestamps from the frames themselves, so this may just be a firmware bug.

dorodnic, 2016-11-19:
A fix has been pushed as part of one of the newer releases, now should work fine.

closed 2016-11-19 by dorodnic