realsenseai/librealsense #203: Video4Linux backend patching

jb892, 2016-07-07:
| Required Info  |   |
|---|---|
| Camera Model | R200 | 
| Firmware Version | 1.0.72.06  | 
| Operating System & Version |  Ubuntu 16.04 LTS |
| Kernel Version (Linux Only)  |  4.4.0-040400-generic |
| Build System |   Makefile | 

```
$ sudo dmesg | tail -n 50
...
[   27.284877] usb 4-1: Product: Intel RealSense 3D Camera R200
[   27.284878] usb 4-1: Manufacturer: Intel Corp
[   27.284879] usb 4-1: SerialNumber: SN_2421007586
[  146.474076] usb 4-1: usbfs: process 2512 (cpp-capture) did not claim interface 3 before use
[  146.477690] usb 4-1: usbfs: process 2512 (cpp-capture) did not claim interface 5 before use
[  152.267237] usb 4-1: usbfs: process 2514 (cpp-capture) did not claim interface 2 before use
[  152.283057] usb 4-1: usbfs: process 2514 (cpp-capture) did not claim interface 4 before use
[  162.653226] usb 4-1: USB disconnect, device number 2
[  597.419219] usb 4-1: new SuperSpeed USB device number 3 using xhci_hcd
[  597.437910] usb 4-1: New USB device found, idVendor=8086, idProduct=0a80
[  597.437916] usb 4-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  597.437920] usb 4-1: Product: Intel RealSense 3D Camera R200
[  597.437923] usb 4-1: Manufacturer: Intel Corp
[  597.437926] usb 4-1: SerialNumber: SN_2421007586
```

ddiakopoulos, 2016-07-14:
Do you have another system to try? I've seen many of the symptoms of patch failures, but this really stands out: `[  162.653226] usb 4-1: USB disconnect, device number 2`. It might be a hardware issue related that particular USB port, or even the USB host controller...

jb892, 2016-07-15:
I have tried on same PC using same USB port but in windows 10. It works fine. However, when I was doing re-patching in Ubuntu. I got this error:
`./patch-uvcvideo-4.4.sh: line 47: ../scripts/realsense-camera-formats.patch: No such file or directory
`
Command:
`$ ./scripts/patch-uvcvideo-4.4.sh v4.4-wily`

jb892, 2016-07-16:
P.S: I got an error even when I run this command:
```
jack@jack-All-Series:~/Documents/librealsense$ ./scripts/install_dependencies-4.4.sh

Installing gcc-4.9

 Toolchain test builds; see https://wiki.ubuntu.com/ToolChain

 More info: https://launchpad.net/~ubuntu-toolchain-r/+archive/ubuntu/test
Press [ENTER] to continue or ctrl-c to cancel adding it

gpg: keyring `/tmp/tmplrr34ap1/secring.gpg' created
gpg: keyring `/tmp/tmplrr34ap1/pubring.gpg' created
gpg: requesting key BA9EF27F from hkp server keyserver.ubuntu.com
gpg: /tmp/tmplrr34ap1/trustdb.gpg: trustdb created
gpg: key BA9EF27F: public key "Launchpad Toolchain builds" imported
gpg: Total number processed: 1
gpg:               imported: 1  (RSA: 1)
OK
Hit:1 http://ppa.launchpad.net/maarten-baert/simplescreenrecorder/ubuntu xenial InRelease
Hit:2 http://archive.canonical.com/ubuntu xenial InRelease                     
Hit:3 http://ppa.launchpad.net/ubuntu-toolchain-r/test/ubuntu xenial InRelease 
Ign:4 http://ppa.launchpad.net/v-launchpad-jochen-sprickerhof-de/pcl/ubuntu xenial InRelease
Ign:5 http://ppa.launchpad.net/v-launchpad-jochen-sprickerhof-de/pcl/ubuntu xenial Release
Hit:6 http://gb.archive.ubuntu.com/ubuntu xenial InRelease                     
Ign:7 http://ppa.launchpad.net/v-launchpad-jochen-sprickerhof-de/pcl/ubuntu xenial/main amd64 Packages
Ign:8 http://ppa.launchpad.net/v-launchpad-jochen-sprickerhof-de/pcl/ubuntu xenial/main i386 Packages
Hit:9 http://gb.archive.ubuntu.com/ubuntu xenial-updates InRelease             
Ign:10 http://ppa.launchpad.net/v-launchpad-jochen-sprickerhof-de/pcl/ubuntu xenial/main all Packages
Ign:11 http://ppa.launchpad.net/v-launchpad-jochen-sprickerhof-de/pcl/ubuntu xenial/main Translation-en_GB
Ign:12 http://ppa.launchpad.net/v-launchpad-jochen-sprickerhof-de/pcl/ubuntu xenial/main Translation-en
Hit:13 http://gb.archive.ubuntu.com/ubuntu xenial-backports InRelease          
Ign:14 http://ppa.launchpad.net/v-launchpad-jochen-sprickerhof-de/pcl/ubuntu xenial/main amd64 DEP-11 Metadata
Ign:15 http://ppa.launchpad.net/v-launchpad-jochen-sprickerhof-de/pcl/ubuntu xenial/main DEP-11 64x64 Icons
Ign:7 http://ppa.launchpad.net/v-launchpad-jochen-sprickerhof-de/pcl/ubuntu xenial/main amd64 Packages
Ign:8 http://ppa.launchpad.net/v-launchpad-jochen-sprickerhof-de/pcl/ubuntu xenial/main i386 Packages
Ign:10 http://ppa.launchpad.net/v-launchpad-jochen-sprickerhof-de/pcl/ubuntu xenial/main all Packages
Ign:11 http://ppa.launchpad.net/v-launchpad-jochen-sprickerhof-de/pcl/ubuntu xenial/main Translation-en_GB
Ign:12 http://ppa.launchpad.net/v-launchpad-jochen-sprickerhof-de/pcl/ubuntu xenial/main Translation-en
Ign:14 http://ppa.launchpad.net/v-launchpad-jochen-sprickerhof-de/pcl/ubuntu xenial/main amd64 DEP-11 Metadata
Ign:15 http://ppa.launchpad.net/v-launchpad-jochen-sprickerhof-de/pcl/ubuntu xenial/main DEP-11 64x64 Icons
Ign:7 http://ppa.launchpad.net/v-launchpad-jochen-sprickerhof-de/pcl/ubuntu xenial/main amd64 Packages
Ign:8 http://ppa.launchpad.net/v-launchpad-jochen-sprickerhof-de/pcl/ubuntu xenial/main i386 Packages
Ign:10 http://ppa.launchpad.net/v-launchpad-jochen-sprickerhof-de/pcl/ubuntu xenial/main all Packages
Ign:11 http://ppa.launchpad.net/v-launchpad-jochen-sprickerhof-de/pcl/ubuntu xenial/main Translation-en_GB
Hit:16 http://packages.ros.org/ros/ubuntu xenial InRelease               
Ign:12 http://ppa.launchpad.net/v-launchpad-jochen-sprickerhof-de/pcl/ubuntu xenial/main Translation-en
Get:17 http://security.ubuntu.com/ubuntu xenial-security InRelease [94.5 kB]
Ign:14 http://ppa.launchpad.net/v-launchpad-jochen-sprickerhof-de/pcl/ubuntu xenial/main amd64 DEP-11 Metadata
Ign:15 http://ppa.launchpad.net/v-launchpad-jochen-sprickerhof-de/pcl/ubuntu xenial/main DEP-11 64x64 Icons
Ign:7 http://ppa.launchpad.net/v-launchpad-jochen-sprickerhof-de/pcl/ubuntu xenial/main amd64 Packages
Ign:8 http://ppa.launchpad.net/v-launchpad-jochen-sprickerhof-de/pcl/ubuntu xenial/main i386 Packages
Ign:10 http://ppa.launchpad.net/v-launchpad-jochen-sprickerhof-de/pcl/ubuntu xenial/main all Packages
Ign:11 http://ppa.launchpad.net/v-launchpad-jochen-sprickerhof-de/pcl/ubuntu xenial/main Translation-en_GB
Ign:12 http://ppa.launchpad.net/v-launchpad-jochen-sprickerhof-de/pcl/ubuntu xenial/main Translation-en
Ign:14 http://ppa.launchpad.net/v-launchpad-jochen-sprickerhof-de/pcl/ubuntu xenial/main amd64 DEP-11 Metadata
Ign:15 http://ppa.launchpad.net/v-launchpad-jochen-sprickerhof-de/pcl/ubuntu xenial/main DEP-11 64x64 Icons
Ign:7 http://ppa.launchpad.net/v-launchpad-jochen-sprickerhof-de/pcl/ubuntu xenial/main amd64 Packages
Ign:8 http://ppa.launchpad.net/v-launchpad-jochen-sprickerhof-de/pcl/ubuntu xenial/main i386 Packages
Ign:10 http://ppa.launchpad.net/v-launchpad-jochen-sprickerhof-de/pcl/ubuntu xenial/main all Packages
Ign:11 http://ppa.launchpad.net/v-launchpad-jochen-sprickerhof-de/pcl/ubuntu xenial/main Translation-en_GB
Ign:12 http://ppa.launchpad.net/v-launchpad-jochen-sprickerhof-de/pcl/ubuntu xenial/main Translation-en
Ign:14 http://ppa.launchpad.net/v-launchpad-jochen-sprickerhof-de/pcl/ubuntu xenial/main amd64 DEP-11 Metadata
Ign:15 http://ppa.launchpad.net/v-launchpad-jochen-sprickerhof-de/pcl/ubuntu xenial/main DEP-11 64x64 Icons
Err:7 http://ppa.launchpad.net/v-launchpad-jochen-sprickerhof-de/pcl/ubuntu xenial/main amd64 Packages
  404  Not Found
Ign:8 http://ppa.launchpad.net/v-launchpad-jochen-sprickerhof-de/pcl/ubuntu xenial/main i386 Packages
[... cut ...]
W: The repository 'http://ppa.launchpad.net/v-launchpad-jochen-sprickerhof-de/pcl/ubuntu xenial Release' does not have a Release file.
N: Data from such a repository can't be authenticated and is therefore potentially dangerous to use.
N: See apt-secure(8) manpage for repository creation and user configuration details.
E: Failed to fetch http://ppa.launchpad.net/v-launchpad-jochen-sprickerhof-de/pcl/ubuntu/dists/xenial/main/binary-amd64/Packages  404  Not Found
E: Some index files failed to download. They have been ignored, or old ones used instead.

```

oordan, 2017-08-01:
This should be no longer an issue with the updated [script](https://github.com/IntelRealSense/librealsense/blob/master/scripts/patch-realsense-ubuntu-xenial.sh).

closed 2017-08-01 by oordan