realsenseai/librealsense #205: Cannot find device at SSH sessiom

ozaslan, 2016-07-10:
| Required Info  |   |
|---|---|
| Camera Model | R200 | 
| Firmware Version |  1.0.72.06 | 
| Operating System & Version | Ubuntu 14.04.4 LTS  |
| Kernel Version (Linux Only)  | 4.2.0-040200-generic |
| Build System |   Makefile | 

Hi,

The driver works properly when I login through Unity. However when I run an example in an SSH session, all the applications such as "cpp-enumerate-devices", give the error "No device detected. Is it plugged in?". It is interesting that if I login through Unity "and" run the binaries in an SSH session, applications work properly. And when I log out from Unity, I again get the same error messages. It 
looks like some start up script is bind to Unity shell. Is there a way to work around this problem?

Thanks

ddiakopoulos, 2016-07-14:
@ozaslan did you make any further progress on debugging the issue? This isn't consistent with my experience with librealsense -- I've gotten it working on a number of headless boards (no display hardware at all) via SSH.

ozaslan, 2016-07-14:
@ddiakopoulos, can you share which Linux distro are you using? For the time being, I made it to work by setting 'Automatic Login'.

ddiakopoulos, 2016-07-16:
@ozaslan interesting! Good to note. We have a number of packages to support embedded distros here (Yocto, Wind River, Ostro): https://github.com/IntelRealSense/meta-intel-librealsense

Tagging this as a bug to help others who might encounter a similar issue but closing for now. If anyone else chimes in with a similar problem, we can re-open and try to root cause.

closed 2016-07-16 by ddiakopoulos