realsenseai/librealsense #208: R200 Realsense Depth does not work outdoors

gabrielarpino, 2016-07-12:
Hi, 

I was able to get my R200 Realsense camera to work, but I noticed that the depth camera does not work well outdoors. However, I tested the R200 realsense with the deprecated Ros Realsense package (https://github.com/PercATI/RealSense_ROS) which is not recommended, and the depth worked very well outdoors in sunlight. Why is this and how might I be able to change the configurations on librealsense to capture correct outdoors depth data?

Thanks.

|---|---|
| Camera Model | R200 | 
| Firmware Version |  1.0.72.06 | 
| Operating System & Version | Linux Ubuntu 14.04 LTS  |
| Kernel Version (Linux Only)  |  4.2.0 |
| Build System |  Makefile |

ddiakopoulos, 2016-07-12:
High-level wrappers like ROS packages tend to pre-configure a lot of settings. As you've noted, R200 works fine outdoors with the right settings. With the cpp-config-ui example in librealsense, I suggest playing with turning IR autoexposure on/off, and also playing with the gain and exposure settings while it is off.

gabrielarpino, 2016-07-12:
Hi ddiakopoulos, I will try playing with those settings and will let you know. Thanks!

ddiakopoulos, 2016-07-16:
@gabrielarpino any closure on your issue?

gabrielarpino, 2016-07-17:
Hi @ddiakopoulos, 

Yes, I found ideal settings for Realsense to be used outdoors with librealsense. I will post the configurations shortly so that other people can get their Realsense working outdoors.

liyuming1978, 2016-07-18:
Hi @gabrielarpino 
     I find r200 depth data will be lost if the object is black, maybe your configure data can work.  Thanks if you can share your data ,

gabrielarpino, 2016-07-24:
Hi @liyuming1978 ,

Here is my text file with the optimal parameters for outdoor realsense usage. You can either save this text into a text file and run rosparam load before running r200_nodelet_rgbd.launch if you're using this, or manually change the configurations based on what is written:

camera:
  base_frame_id: camera_link
  camera: {base_frame_id: camera_link, color_backlight_compensation: 0, color_brightness: 62,
    color_contrast: 38, color_enable_auto_white_balance: 0, color_fps: 30, color_frame_id: camera_rgb_frame,
    color_gain: 88, color_gamma: 180, color_height: 480, color_hue: -1018, color_optical_frame_id: camera_rgb_optical_frame,
    color_saturation: 87, color_sharpness: 7, color_white_balance: 2987, color_width: 640,
    depth_fps: 30, depth_frame_id: camera_depth_frame, depth_height: 360, depth_optical_frame_id: camera_depth_optical_frame,
    depth_width: 480, enable_color: true, enable_depth: true, enable_pointcloud: true,
    enable_tf: true, ir2_frame_id: camera_ir2_frame, ir_frame_id: camera_ir_frame,
    mode: manual, r200_auto_exposure_bottom_edge: 239, r200_auto_exposure_left_edge: 0,
    r200_auto_exposure_right_edge: 319, r200_auto_exposure_top_edge: 0, r200_emitter_enabled: 1,
    r200_lr_auto_exposure_enabled: 1, r200_lr_exposure: 7, r200_lr_gain: 100, serial_no: '',
    usb_port_id: ''}
  camera_nodelet_manager: {num_worker_threads: 4}
  color:
    image_raw:
      compressed: {format: jpeg, jpeg_quality: 80, png_level: 9}
      compressedDepth: {depth_max: 10.0, depth_quantization: 100.0, png_level: 9}
      theora: {keyframe_frequency: 64, optimize_for: 1, quality: 31, target_bitrate: 800000}
  color_backlight_compensation: 0
  color_brightness: 62
  color_contrast: 38
  color_enable_auto_white_balance: 0
  color_gain: 88
  color_gamma: 180
  color_hue: -1018
  color_optical_frame_id: camera_rgb_optical_frame
  color_saturation: 87
  color_sharpness: 7
  color_white_balance: 2987
  r200_auto_exposure_bottom_edge: 239
  r200_auto_exposure_left_edge: 0
  r200_auto_exposure_right_edge: 319
  r200_auto_exposure_top_edge: 0
  r200_emitter_enabled: 1
  r200_lr_auto_exposure_enabled: 1
  r200_lr_exposure: 7
  r200_lr_gain: 100

ddiakopoulos, 2016-07-31:
Closing this out to clean up the issue queue. Thanks for your work in documenting the correct settings! Ping @mdhorn who might want to add this as convenient documentation to the ROS wrapper.

closed 2016-07-31 by ddiakopoulos

gabrielarpino, 2016-08-01:
Sounds good, thanks!

mdhorn, 2016-08-01:
I captured a story to generate a sample launch file using these settings in our backlog.

ChanghyeonKim93, 2016-10-28:
@ddiakopoulos 
Hi, I'm a user of intel realsense R200. 
Thanks for your answering, I can set the realsense parameters by running "cpp-config-ui.cpp".
But, parameter settings just are valid only when "cpp-config-ui.cpp" 's GUI. So I cannot use the set parameters on my own c++ code applications . 
I analysed the source code of the "cpp-config-ui.cpp", but I cannot find the way to set realsense params on my code. 
Could you tell me the way to use the parameter settings on my code? or I have to make cmake file on some specific way? 
Thanks. 

ps. I use the Linux 14.04 LTS .