realsenseai/librealsense #209: run example on Ubuntu，IR&depth image wrong while color image correct

zhouxiangming18, 2016-07-14:
| Required Info  |   |
|---|---|
| Camera Model | F200 | 
| Firmware Version | 1.4.27.52404  | 
| Operating System & Version | Ubuntu 16.04 x64  & Ubuntu 14.04 x64  |
| Kernel Version (Linux Only)  |  4.4.0-21 |
| Build System |  Makefile | 
[image]

| Required Info  |   |
|---|---|
| Camera Model | SR300 | 
| Firmware Version | 3.1.25.2599  | 
| Operating System & Version | Ubuntu 16.04 x64  & Ubuntu 14.04 x64|
| Kernel Version (Linux Only)  |  4.4.0-21 |
| Build System |  Makefile | 
[image]

ddiakopoulos, 2016-07-14:
Hi @zhouxiangming18. These are common indications of an installation issue where our script didn't properly patch the `uvcvideo.ko` driver. I recommend re-running the patch script for your configuration (and also remember to execute `sudo modprobe uvcvideo` when finished). Please note any issues or error messages encountered during the patch process and post them here, if relevant. 

Other helpful debugging tips:
1. Make sure the camera is unplugged during the patch script
2. After finishing and running `sudo modprobe uvcvideo`, execute `dmesg` after plugging in the camera and see if there are any immediate errors within the past 10-20 lines

ddiakopoulos, 2016-07-31:
Closed for lack of new info.

closed 2016-07-31 by ddiakopoulos