realsenseai/librealsense #210: F200 Infrared stream can't get with OpenCV

harumo11, 2016-07-14:
| Required Info  |   |
|---|---|
| Camera Model | R200, SR300 | 
| Firmware Version |  1.0.72.06, 3.10.10.0 | 
| Operating System & Version | Ubuntu 16.04  |
| Kernel Version (Linux Only)  | 4.4.0-28-generic  |
| Build System |  Makefile | 

Hello and Thanks for good library.

I tried to get infrared stream and convert OpenCV Mat with realsense R200.
But a following error happened. 

```
terminate called after throwing an instance of 'rs::error'
  what():  Timeout waiting for frames.
Aborted (core dumped)
```

I have a realsense SR300 too. So I tried using same program. It's worked well.

Do you have any ideas?
Thanks.

ddiakopoulos, 2016-07-31:
@harumo11 I would try your R200 with a different physical USB port or through a powered USB 3.0 hub. The bandwidth/power requirements vary between generations/families of RealSense cameras.

harumo11, 2016-08-09:
@ddiakopoulos Hello. I tried my R200 camera with another physical USB port. But same error has happened. I tried with a following code.  Do I make any mistakes? 

```
// This program is handling intel realsense SR300 with infrared image.
// Note. This program is only used on SR300, con not used on f200 due to unknoun issue.

#include <iostream>
#include <opencv2/core.hpp>
#include <opencv2/highgui.hpp>
#include "librealsense/rs.hpp"

using std::cout;
using std::endl;

int main(int argc, char const* argv[])
{
	rs::context ctx;
	cout << "[ INFO ] connected " << ctx.get_device_count() 
		<< "realsense deveces" << endl;
	if (ctx.get_device_count() == 0) {
		cout << "[ ERROR ] No devece is connected" << endl;
		return 0;
	}

	rs::device *dev = ctx.get_device(0);
	cout << "------------------------------------------" << endl;
	cout << "[ INFO ] Devece name\t" << dev->get_name() << endl;
	cout << "[ INFO ] Serial number\t" << dev->get_serial() << endl;
	cout << "[ INFO ] Firmware version\t" << dev->get_firmware_version() << endl;
	cout << "------------------------------------------" << endl;

	// Setting for realsense device
	const int frameHeight = 480;
	const int frameWidth = 640;
	dev->enable_stream(rs::stream::infrared, frameWidth, frameHeight, rs::format::y8, 30);
	dev->start();

	// Setting for openCV
	cv::Mat infrared(frameHeight, frameWidth, CV_8UC1); 

	while(cv::waitKey(30) <= 0){
		dev->wait_for_frames();
		if (!dev->is_streaming()) {
			cout << "[ ERROR ] Disable streaming  " << endl;
		}

		if (!dev->is_stream_enabled(rs::stream::infrared)) {
			cout << "[ ERROR ] Disable infrared stream" << endl;
			return 1;
		}

		infrared.data = (unsigned char*)dev->get_frame_data(rs::stream::infrared);
		cv::imshow("infrared", infrared);
	}
	
	return 0;
}
```

Thanks.

dorodnic, 2016-08-09:
Please try adding:
`try { dev->enable_stream(rs::stream::infrared2, frameWidth, frameHeight, rs::format::y8, 30); } catch(...) {}` after the first enable_stream. 
It seemed that Y8 format used for DS4 single infrared streaming has not been up-streamed to kernel 4.4 nor added to librealsense installation script. 
You can either enable both left and right infrared streams or manually apply the following patch:
https://github.com/IntelRealSense/librealsense/blob/development/scripts/realsense-y8-format-request.patch

harumo11, 2016-08-09:
@dorodnic  Hello.
I tried what you say.  I can get infrared stream after changing infrared camera from rs::stream::infrared to rs::stream::infrared2. And then,  Following code work well.

```
// This program is handling intel realsense SR300 and R200 with infrared image.
// Note. This program surely is used on SR300 and R200.

#include <iostream>
#include <opencv2/core.hpp>
#include <opencv2/highgui.hpp>
#include "librealsense/rs.hpp"

using std::cout;
using std::endl;

int main(int argc, char const* argv[])
{
	rs::context ctx;
	cout << "[ INFO ] connected " << ctx.get_device_count() 
		<< "realsense deveces" << endl;
	if (ctx.get_device_count() == 0) {
		cout << "[ ERROR ] No devece is connected" << endl;
		return 0;
	}

	rs::device *dev = ctx.get_device(0);
	cout << "------------------------------------------" << endl;
	cout << "[ INFO ] Devece name\t" << dev->get_name() << endl;
	cout << "[ INFO ] Serial number\t" << dev->get_serial() << endl;
	cout << "[ INFO ] Firmware version\t" << dev->get_firmware_version() << endl;
	cout << "------------------------------------------" << endl;

	// Setting for realsense device
	const int frameHeight = 480;
	const int frameWidth = 640;

	try{
		dev->enable_stream(rs::stream::infrared2, frameWidth, frameHeight, rs::format::y8, 30);
	}catch(...){
	}

	dev->start();

	// Setting for openCV
	cv::Mat infrared(frameHeight, frameWidth, CV_8UC1); 

	while(cv::waitKey(30) <= 0){
		dev->wait_for_frames();
		if (!dev->is_streaming()) {
			cout << "[ ERROR ] Disable streaming  " << endl;
		}

		if (!dev->is_stream_enabled(rs::stream::infrared2)) {
			cout << "[ ERROR ] Disable infrared stream" << endl;
			return 1;
		}

		infrared.data = (unsigned char*)dev->get_frame_data(rs::stream::infrared2);
		cv::imshow("infrared", infrared);
	}
	
	return 0;
}
```
Thank you so much.

harumo11, 2016-08-09:
I could get infrared image with R200 and SR300.
So problem is solved. Then I will close this issue.

Thanks good library.

closed 2016-08-09 by harumo11