realsenseai/librealsense #213: Screen tearing on sample applications

neale, 2016-07-14:
| Required Info  |   |
|---|---|
| Camera Model | SR300 | 
| Firmware Version |   | 
| Operating System & Version | Ubuntu 14.04  |
| Kernel Version (Linux Only)  | 4.4.0-040400-generic  |
| Build System | Makefile | 

I get massive screen tearing using the example applications with the camera. The camera displays the correct views, but there is both lag, and noise. 
When loading `cpp-capture` I get:
```
Capturing DEPTH at 640 x 480, fox = 67.9 x 53.6, distortion = INVERSE_BROWN_CONRADY
Capturing COLOR at 640 x 480, fox = 54.8 x 42.5, distortion = NONE
Capturing INFRARED at 640 x 480, fox = 67.9 x 53.6, distortion = INVERSE_BROWN_CONRADY
libEGL warning: DRI2: failed to authenticate
```
Using `c-depth` binary that gives an ascii depth view, I get the same horizontal lines reminiscent of screen tearing. 
Rendering a windows with OpenCV that simply uses the camera as a webcam gives no issues. 
I've done the install process multiple times and no resolution. 

I've also installed `libgles2-mesa-dev` and `x11proto-dri2-dev` if that helps

ddiakopoulos, 2016-07-14:
Hi @neale -- we use GLFW in its most simple form with no fancy modern OpenGL features. Unfortunately this issue seems deeper than an issue with this library (probably display driver related...). Just a quick google of `libEGL warning: DRI2: failed to authenticate` shows tons of system config related issues.

neale, 2016-07-14:
It looks that way, it seems to happen to Ubuntu users when they have libEGL files installed somewhere other than /usr/lib/x86_64-linux-gnu/mesa-egl/

neale, 2016-07-14:
Ill close this and reopen if I know definitively that the problem is in librealsense

closed 2016-07-14 by neale