realsenseai/librealsense #214: SR300 color image noise

mdhankhar, 2016-07-22:
| Required Info  |   |
|---|---|
| Camera Model | SR300 | 
| Firmware Version | 3.15.0.0  | 
| Operating System & Version | Ubuntu 16.04  |
| Kernel Version (Linux Only)  | 4.4.0-31-generic  |
| Build System |  Makefile | 

Hi,

I'm trying to get SR300 to work on a fresh Ubuntu install. But color image keeps flickering (video: https://goo.gl/photos/59RHiTFQLkwywxkT8). Here librealsense has LibUVC backend. I also tried separately with v4l backend, in which case the depth image looked messed up.

(BTW, everything works fine on MacOS / Windows.)

Any suggestions to fix the issue? Thanks.

dorodnic, 2016-07-25:
Are you running Ubuntu in a virtual machine or a slow host? 
It does not look like a regular kernel-patch issue (although please make sure you follow the instructions in [installation guide](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md), not all SR300 stream formats has been up-streamed so the camera will not work out-of-the-box without a patch).
We did encounter something similar when working on a VM. To resolve it, you change #define UVC_URBS   4 to #define UVC_URBS  32 in /drivers/media/usb/uvc/uvc_video.h and rebuild the kernel.
You can give it a try and let us know if it helped.

mdhankhar, 2016-07-25:
thanks for the suggestion, but it already worked right away on Ubuntu 14.04 with the patch script mention in the Installation guide (./scripts/patch-uvcvideo-ubuntu-mainline.sh)

ddiakopoulos, 2016-07-31:
@mdhankhar you should prefer using the V4L2 backend... it seems like all the steps in the `installation.md` guide weren't followed if you saw an incorrect depth image. The issues you see with the LibUVC backend are with a pre-configured buffer size (can't remember if they need to be larger or smaller for SR300) in libuvc and don't occur with V4L2.

jlgarcia75, 2017-10-06:
For best support, post issues in the [RealSense Community](https://communities.intel.com/community/tech/realsense).

Hello,

Do you still need help with this issue?

Best regards,
Jesus
Intel Customer Support

ddiakopoulos, 2017-10-07:
Hi @jlgarcia75 -- i think for best support, issues should be responded directly in the issue queue for this repository.