realsenseai/librealsense #215: R200 does not work anymore after last commits

nlyubova, 2016-07-22:
| Required Info  |   |
| Camera Model | R200 | 
| Firmware Version | 1.0.72.04  | 
| Operating System & Version | Ubuntu 14.04  |
| Kernel Version (Linux Only)  | 4.4.0-040400-generic  |
| Build System |  Makefile | 

Hello,
after the last commits, our R200 does not work anymore... The log is attached. However, the camera still works with the older version of the code. Is it possible to fix it? 
 
ba$ ./bin/cpp-capture 
rs.warn: Camera head contents version != 12, data may be missing/incorrect
*** Error in `./bin/cpp-capture': corrupted double-linked list: 0x0000000002664e30 ***
Aborted (core dumped)

ddiakopoulos, 2016-07-22:
@dorodnic can your team look at this ASAP?

ddiakopoulos, 2016-07-22:
@mdhorn can the ROS team replicate this issue?

mdhorn, 2016-07-22:
@ddiakopoulos This is makefile problem. The latest updates doesn't account for all file dependencies.

The problem reported can be reproduced by:

- checkout previous working version (say the v0.9.2 tag)
- make clean
- make
- sudo make install
- checkout master
- make 
- make install

If you do a 'make clean' after checking out the HEAD of master before the 'make' and 'sudo make install', the problem does not occur.

The latest change to the Makefile doesn't account for all dependencies so not all required files are rebuilt from sources.

dorodnic, 2016-07-23:
There has been a lot of small changes to the makefile lately, we will localize and resolve shortly.
As Mark has mentioned, there is a simple work around. However, if necessary you can always fall back to 0.9.1.

nlyubova, 2016-07-23:
The code works for SR300 but not for R200

dorodnic, 2016-07-24:
Should be resolved now.
Please retry with latest sources and let us know.

nlyubova, 2016-07-25:
Thank you! now it works again !

closed 2016-07-25 by nlyubova