realsenseai/librealsense #216: R200 failing on librealsense

championoaks, 2016-07-22:
| Required Info  |   |
|---|---|
| Camera Model | R200, F200 or SR300 | R200
| Firmware Version | Firmware version:	1.0.72.06| 
| Operating System & Version |  Windows 10 RS1 |
| Kernel Version (Linux Only)  |   |
| Build System |  VS2013, XCode or Makefile | VS2013

Firmware version:	1.0.72.06
RealSense error calling rs_create_context(api_version:4):
    IKsControl::KsProperty(...) returned 0x80004003

ddiakopoulos, 2016-07-22:
R200 support seems to be broken on the master branch. Can you use an older tagged release? https://github.com/IntelRealSense/librealsense/releases/tag/v0.9.1

Ping @dorodnic

dorodnic, 2016-07-23:
@championoaks you are trying to work the librealsense using Windows 10 Red-Stone pre-release, correct? Could you please specify the exact build you are using. 
Also please let us know if rolling back to previous version changed anything. 
There are major changes in Media Foundation architecture in Red-Stone (introduction of the frame-server and multi-app support). 
@ddiakopoulos can you confirm that R200 is not working on Windows? I have tested it a week 
 ago to work just fine.

dorodnic, 2016-07-23:
As a work-around, you can try to disable the Frame-Server Red-Stone feature, by modifying the following registry keys: 
[HKEY_LOCAL_MACHINE\SOFTWARE\Microsoft\Windows Media Foundation\Platform]
"EnableFrameServerMode"=dword:00000000 
[HKEY_LOCAL_MACHINE\SOFTWARE\Wow6432Node\Microsoft\Windows Media Foundation\Platform]
"EnableFrameServerMode"=dword:00000000

championoaks, 2016-07-24:
I don't think its Frameserver related as SR300 works fine. OS Build # 14393

ddiakopoulos, 2016-07-31:
@championoaks can you try again with the last master branch?

dorodnic, 2016-08-01:
I was able to reproduce this exact behavior with build 14393. Everything is working fine with Frame-Server disabled. Seem like new RS issue with Media Foundation backward compatibility. 
@championoaks Closing the issue based on the private conversation we had.

closed 2016-08-01 by dorodnic

Maghoumi, 2016-08-03:
The workaround does not seem to work for me on Windows 10 after the Anniversary Update today.