realsenseai/librealsense #218: Data format of streams

francescoli, 2016-07-25:
| Required Info  |   |
|---|---|
| Camera Model | R200 | 
| Firmware Version | 1.0.72.06 | 
| Operating System & Version | Ubuntu 14.04.4 |
| Kernel Version (Linux Only)  |  4.4.0-040400-generic |
| Build System | Makefile | 

Hi, @ddiakopoulos 
I've noticed that there are several synthetic streams available. I wonder how the data of the  color_aligned_to_depth stream organized, maybe just like rgb,rgb,rgb...rgb,d,d,d...d? And is this data suitable for direct use of camera tracking and mapping? In freenect2, I find that a registration is required between depth and color streams.
Another question, the coeffs array of struct rs_intrinsics is organized radial distortion first or tangential?
Regards,

francescoli, 2016-07-25:
Oh, I have made a mistake of the concept of depth_aligned_to_color, it only cotains depth data.
In the examples, I just find enbale_stream of color and depth, no depth_aligned_to_color, only assertion of is_stream_depth. Does it means just enable the native streams make all works?

ddiakopoulos, 2016-07-31:
Correct. The _synthetic streams_ are always available if you enable the correct _native streams_. Per your original question, note the documentation in the header file that specifies the intent of the synthetic streams (notably that depth_aligned_to_color would be a depth map with the intrinsics of the RGB image, and color_aligned_to_depth would be a color image with the intrinsics of the depth map).

closed 2016-07-31 by ddiakopoulos

l3s777, 2016-08-11:
Hi!
I am opening the streams for color, depth, infrared and I want to use  color_aligned_to_depth to show that image result in a matrix in openCV. Any hint how to use it?
An example about who I am doing librealsense + openCV in color:
cv::Mat rgb_show;
const uint8_t * rgb_frame = reinterpret_cast<const uint8_t*(dev->get_frame_data(rs::stream::color));
cv::Mat rgb = cv::Mat(480, 640, CV_8UC3, (void*)rgb_frame);
cvtColor(rgb, rgb_show, CV_BGR2RGB);
cv::imshow("RGB", rgb_show);

thanks!