realsenseai/librealsense #220: Incorrect usb-port-id for F200 cameras

reaganlo, 2016-07-25:
| Required Info  |   |
|---|---|
| Camera Model | **F200** | 
| Firmware Version |  2.60.0.0 | 
| Operating System & Version | Ubuntu 16.04.1 LTS   |
| Kernel Version (Linux Only)  |  4.4.0-31-generic |
| Build System |  Makefile | 
| librealsense | v0.9.2 |

For **F200** cameras, the usb-port-id always returns the same value even if the camera is connected to different USB ports.
**This works correctly for R200 cameras**.

Steps to reproduce:
- Plug an F200 camera to a USB port. Run ./cpp-enumerate-devices example and note the value of USB Port ID that gets printed.
- Unplug and plug the F200 camera to a different USB port and repeat the above steps.

mdhorn, 2016-07-26:
The same is true for the SR300 using the same OS and kernel.
SR300: Firmware version: 3.10.10.0
Always returns the same value.

dorodnic, 2016-08-10:
I have been trying to reproduce this for some time now with Ubuntu 14.04 and 16.04 with 4.4.0-34-generic, F200 and SR300 but I don't see this behavior. Do you localize it to a specific kernel / setup? What is the result being returned?

mdhorn, 2016-08-11:
We can reproduce this on both 14.04 and 16.04.

Using 16.04 system with 4.4.0-34-generic, attach 2 or more SR300 cameras.
```
cpp-enumerate-devices | grep -i usb
 USB Port ID: 4-1
 USB Port ID: 4-1
 USB Port ID: 4-1
```
All three device report the same USB Port ID.

I'm attaching the full output of SR300 commands `cpp-enumerate-devices` as [cpp_enumerate_devices.txt](https://github.com/IntelRealSense/librealsense/files/412386/cpp_enumerate_devices.txt) and also the full output of `lsusb -t` as [lsusb__t.txt](https://github.com/IntelRealSense/librealsense/files/412385/lsusb__t.txt) and `lsusb -v` as [lsusb__v.txt](https://github.com/IntelRealSense/librealsense/files/412384/lsusb__v.txt).

We also see the same for SR300 under ubuntu 14.04 with a 4.4.0-34-generic kernel:
```
./bin/cpp-enumerate-devices  | grep -i "usb port"
 USB Port ID: 2-1
 USB Port ID: 2-1
```

However, if we attached multiple R200 cameras:
```
cpp-enumerate-devices | grep -i usb
 USB Port ID: 4-2
 USB Port ID: 4-1
```
I'm attaching the full output of R200 commands as `cpp-enumerate-devices` as [r200_cpp_enumerate_devices.txt](https://github.com/IntelRealSense/librealsense/files/412400/r200_cpp_enumerate_devices.txt) and also the full output of `lsusb -t` as [r200_lsusb__t.txt](https://github.com/IntelRealSense/librealsense/files/412398/r200_lsusb__t.txt) and `lsusb -v` as [r200_lsusb__v.txt](https://github.com/IntelRealSense/librealsense/files/412399/r200_lsusb__v.txt).

We have also found that we can not call rs_create_context from two different processes for SR300/F200 camera in our ROS node. We get a USB error:
```
libusb_claim_interface(...) returned LIBUSB_ERROR_BUSY
```
Is it possible that is problem could be related to the fact that all devices think they are the same USB Port ID?
We don't see this rs_create_context problem with R200 camera, which correctly report unique USB Port IDs.

dorodnic, 2016-08-11:
Thanks, I'll continue to dig into it

dorodnic, 2016-11-09:
@reaganlo, could you give it a try and update the issue please?

mdhorn, 2016-11-09:
Pull Request #348 results in the following:

```
Device 0 - Intel RealSense SR300:
 Serial number: 60920300464 
 Firmware version: 3.10.10.0
 USB Port ID: 4-2-1
Device 1 - Intel RealSense SR300:
 Serial number: 609203000751
 Firmware version: 3.10.10.0
 USB Port ID: 4-5-2-1
Device 2 - Intel RealSense R200:
 Serial number: 2151000084
 Firmware version: 1.0.72.06
 USB Port ID: 4-1
Device 3 - Intel RealSense F200:
 Serial number: 042140050404
 Firmware version: 2.60.0.0
 USB Port ID: 4-5-3-1
```

dorodnic, 2016-11-09:
These strings are unique per setup and not unit specific like the serial number. I don't see any obvious way to flatten these IDs. 
Is it acceptable for your needs?

mdhorn, 2016-11-09:
The IDs can't be flatten if we are properly addressing them, so this is a correct change.

closed 2016-12-15 by dorodnic