realsenseai/librealsense #221: Unknown pixelformat, uvcvideo: Non-zero status (-71) in video completion handler., and video issues

NP-Hardass, 2016-07-26:
| Required Info  |   |
|---|---|
| Camera Model | F200  | 
| Firmware Version |  2.60.0.0  | 
| Operating System & Version |  Gentoo  |
| Kernel Version (Linux Only)  |  4.4.6/4.4.15  |
| Build System |  Makefile | 

I'm currently working on packaging for my laptop (Ideapad Y700 15ISK) and for Gentoo.
Kernel sources were auto-patched by putting realsense-camera-formats.patch in /etc/portage/patches/sys-kernel/gentoo-sources.
System information (emerge --info):  [https://bpaste.net/raw/5a0fe9be337b](url)
Here is the current Gentoo packaging (if it helps): [https://github.com/NP-Hardass/np-hardass-overlay/tree/master/media-libs/librealsense](url)
Whenever I attempt to use the camera with hangouts (chromium) or guvcview, I get the following dump in dmesg
`
[  144.354162] ------------[ cut here ]------------
[  144.354180] WARNING: CPU: 4 PID: 4885 at drivers/media/v4l2-core/v4l2-ioctl.c:1272 v4l_enum_fmt+0xd72/0x10e0 [videodev]()
[  144.354183] Unknown pixelformat 0x5a564e49
[  144.354185] Modules linked in: libafs(PO) vboxnetflt(O) vboxnetadp(O) vboxdrv(O) vhba(O) bbswitch(O) zram uvcvideo videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_core v4l2_common videodev media snd_hda_codec_hdmi iwlmvm snd_hda_codec_realtek snd_hda_codec_generic btusb btintel kvm_intel iwlwifi snd_hda_intel snd_hda_codec snd_hwdep snd_hda_core ideapad_laptop [last unloaded: nvidia]
[  144.354227] CPU: 4 PID: 4885 Comm: guvcview Tainted: P           O    4.4.15-gentoo #1
[  144.354230] Hardware name: LENOVO 80NV/Allsparks 5A, BIOS CDCN35WW 03/29/2016
[  144.354233]  0000000000000286 0000000028ec59ad ffff880855f43be0 ffffffff8166e798
[  144.354239]  ffff880855f43c28 ffffffffa0d32088 ffff880855f43c18 ffffffff81124a01
[  144.354245]  ffff880855f43da0 0000000000000000 0000000000000001 ffff88085be1e800
[  144.354250] Call Trace:
[  144.354258]  [<ffffffff8166e798>] dump_stack+0x4d/0x65
[  144.354266]  [<ffffffff81124a01>] warn_slowpath_common+0x81/0xc0
[  144.354270]  [<ffffffff81124a97>] warn_slowpath_fmt+0x57/0x70
[  144.354279]  [<ffffffffa0dbd518>] ? uvc_ioctl_enum_fmt+0x98/0xb0 [uvcvideo]
[  144.354288]  [<ffffffffa0d1d712>] v4l_enum_fmt+0xd72/0x10e0 [videodev]
[  144.354295]  [<ffffffffa0d1e3ff>] ? v4l_g_parm+0x4f/0xd0 [videodev]
[  144.354304]  [<ffffffffa0d1dd05>] __video_do_ioctl+0x285/0x320 [videodev]
[  144.354311]  [<ffffffff81201836>] ? release_pages+0xd6/0x280
[  144.354316]  [<ffffffff8124a954>] ? __kmalloc+0x1b4/0x250
[  144.354324]  [<ffffffffa0d1f91f>] video_usercopy+0x33f/0x5b0 [videodev]
[  144.354332]  [<ffffffffa0d1da80>] ? v4l_enum_fmt+0x10e0/0x10e0 [videodev]
[  144.354340]  [<ffffffffa0d1fba0>] video_ioctl2+0x10/0x20 [videodev]
[  144.354347]  [<ffffffffa0d19660>] v4l2_ioctl+0xb0/0xe0 [videodev]
[  144.354353]  [<ffffffff81272e5f>] do_vfs_ioctl+0x2bf/0x490
[  144.354359]  [<ffffffff8113fc4f>] ? task_work_run+0x7f/0xa0
[  144.354366]  [<ffffffff815e0f3e>] ? security_file_ioctl+0x3e/0x60
[  144.354371]  [<ffffffff812730a4>] SyS_ioctl+0x74/0x80
[  144.354378]  [<ffffffff81cdbc57>] entry_SYSCALL_64_fastpath+0x12/0x6a
[  144.354382] ---[ end trace 97ff2b00a0cd119a ]---
[  144.354386] ------------[ cut here ]------------
[  144.354395] WARNING: CPU: 4 PID: 4885 at drivers/media/v4l2-core/v4l2-ioctl.c:1272 v4l_enum_fmt+0xd72/0x10e0 [videodev]()
[  144.354398] Unknown pixelformat 0x495a4e49
[  144.354399] Modules linked in: libafs(PO) vboxnetflt(O) vboxnetadp(O) vboxdrv(O) vhba(O) bbswitch(O) zram uvcvideo videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_core v4l2_common videodev media snd_hda_codec_hdmi iwlmvm snd_hda_codec_realtek snd_hda_codec_generic btusb btintel kvm_intel iwlwifi snd_hda_intel snd_hda_codec snd_hwdep snd_hda_core ideapad_laptop [last unloaded: nvidia]
[  144.354431] CPU: 4 PID: 4885 Comm: guvcview Tainted: P        W  O    4.4.15-gentoo #1
[  144.354433] Hardware name: LENOVO 80NV/Allsparks 5A, BIOS CDCN35WW 03/29/2016
[  144.354436]  0000000000000286 0000000028ec59ad ffff880855f43be0 ffffffff8166e798
[  144.354441]  ffff880855f43c28 ffffffffa0d32088 ffff880855f43c18 ffffffff81124a01
[  144.354446]  ffff880855f43da0 0000000000000000 0000000000000001 ffff88085be1e800
[  144.354451] Call Trace:
[  144.354456]  [<ffffffff8166e798>] dump_stack+0x4d/0x65
[  144.354461]  [<ffffffff81124a01>] warn_slowpath_common+0x81/0xc0
[  144.354465]  [<ffffffff81124a97>] warn_slowpath_fmt+0x57/0x70
[  144.354472]  [<ffffffffa0dbd518>] ? uvc_ioctl_enum_fmt+0x98/0xb0 [uvcvideo]
[  144.354480]  [<ffffffffa0d1d712>] v4l_enum_fmt+0xd72/0x10e0 [videodev]
[  144.354487]  [<ffffffffa0d1e3ff>] ? v4l_g_parm+0x4f/0xd0 [videodev]
[  144.354495]  [<ffffffffa0d1dd05>] __video_do_ioctl+0x285/0x320 [videodev]
[  144.354500]  [<ffffffff81201836>] ? release_pages+0xd6/0x280
[  144.354505]  [<ffffffff8124a954>] ? __kmalloc+0x1b4/0x250
[  144.354512]  [<ffffffffa0d1f91f>] video_usercopy+0x33f/0x5b0 [videodev]
[  144.354520]  [<ffffffffa0d1da80>] ? v4l_enum_fmt+0x10e0/0x10e0 [videodev]
[  144.354528]  [<ffffffffa0d1fba0>] video_ioctl2+0x10/0x20 [videodev]
[  144.354535]  [<ffffffffa0d19660>] v4l2_ioctl+0xb0/0xe0 [videodev]
[  144.354539]  [<ffffffff81272e5f>] do_vfs_ioctl+0x2bf/0x490
[  144.354545]  [<ffffffff8113fc4f>] ? task_work_run+0x7f/0xa0
[  144.354551]  [<ffffffff815e0f3e>] ? security_file_ioctl+0x3e/0x60
[  144.354555]  [<ffffffff812730a4>] SyS_ioctl+0x74/0x80
[... 105 lines cut ...]
[  144.355203]  [<ffffffffa0d1dd05>] __video_do_ioctl+0x285/0x320 [videodev]
[  144.355208]  [<ffffffff81201836>] ? release_pages+0xd6/0x280
[  144.355212]  [<ffffffff8124a954>] ? __kmalloc+0x1b4/0x250
[  144.355220]  [<ffffffffa0d1f91f>] video_usercopy+0x33f/0x5b0 [videodev]
[  144.355227]  [<ffffffffa0d1da80>] ? v4l_enum_fmt+0x10e0/0x10e0 [videodev]
[  144.355235]  [<ffffffffa0d1fba0>] video_ioctl2+0x10/0x20 [videodev]
[  144.355242]  [<ffffffffa0d19660>] v4l2_ioctl+0xb0/0xe0 [videodev]
[  144.355246]  [<ffffffff81272e5f>] do_vfs_ioctl+0x2bf/0x490
[  144.355251]  [<ffffffff8113fc4f>] ? task_work_run+0x7f/0xa0
[  144.355257]  [<ffffffff815e0f3e>] ? security_file_ioctl+0x3e/0x60
[  144.355261]  [<ffffffff812730a4>] SyS_ioctl+0x74/0x80
[  144.355267]  [<ffffffff81cdbc57>] entry_SYSCALL_64_fastpath+0x12/0x6a
[  144.355270] ---[ end trace 97ff2b00a0cd119f ]---
[  152.933594] uvcvideo: Non-zero status (-71) in video completion handler.
[  219.376367] uvcvideo: Non-zero status (-71) in video completion handler.
`
When I use cpp-capture, I get just
`
[ 1207.113442] uvcvideo: Non-zero status (-71) in video completion handler.
[ 1207.119551] uvcvideo: Non-zero status (-71) in video completion handler.
`

Not sure if it is related, but I get no output in guvcview,
I get this in hangouts [image]
and normal output from cpp-capture [image]

ddiakopoulos, 2016-07-31:
Hi @NP-Hardass. Gentoo isn't a validated/tested platform by Intel (but great to see it working with cpp-capture at least!). Using RealSense cameras in guvcview or other system applications aren't supported on Linux -- it looks like it's trying to pull the default depth stream in hangouts and using an RGB pixel format. But, from your screenshot... seems to be working well with cpp-capture, so I don't think there's any action we can take to fix these application-level issues on your system.

closed 2016-07-31 by ddiakopoulos

NP-Hardass, 2016-07-31:
@ddiakopoulos Ah, that must be a misunderstanding on my part.  I thought part of the purpose of the uvcvideo patching was to allow usage in system applications.    Do you by chance know if support by system applications is on the TODO list?  I wonder if it looks like hangouts is pulling the wrong stream, if there is a means of swapping stream orders or the default stream.

Also, as an official Gentoo developer, if there is anything that you need from me to validate our platform, let me know.  I haven't had the opportunity yet, but intend to add support for testing the software with the unit tests in our packaging

Thanks for taking the time to look into the situation.