realsenseai/librealsense #223: R200 does not work well with Skylake

robertxchen66, 2016-07-28:
| Required Info  |   |
|---|---|
| Camera Model | R200 | 
| Firmware Version |   | 
| Operating System & Version |  Ubuntu 14.04.3 |
| Kernel Version (Linux Only)  | 4.4  |
| Build System |  Makefile | 
Hardware: NUC6i5SYB, 16GB

After doing proper uvcvideo driver patch, I can run the sample programs such as cpp-capture and cpp-alignimages. However, the videos are jittery.  And after a while, the system crashs and go to the desktop login screen. Check the dmesg, here is what I got:

[ 4610.981149] drm/i915: Resetting chip after gpu hang
[ 4610.982631] ------------[ cut here ]------------
[ 4610.982664] WARNING: CPU: 3 PID: 5039 at /home/kernel/COD/linux/drivers/gpu/drm/i915/intel_lrc.c:702 intel_logical_ring_begin+0x1ab/0x250 [i915]()
[ 4610.982666] WARN_ON(&target->list == &ring->request_list)
[ 4610.982668] Modules linked in: uvcvideo(OE) videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_core v4l2_common videodev media drbg ansi_cprng ctr ccm snd_hda_codec_hdmi snd_soc_skl snd_soc_skl_ipc snd_hda_ext_core snd_hda_codec_realtek snd_soc_sst_ipc snd_hda_codec_generic snd_soc_sst_dsp snd_soc_core snd_compress ac97_bus snd_pcm_dmaengine dw_dmac_core input_leds snd_hda_intel snd_hda_codec snd_hda_core snd_hwdep hid_generic 8250_dw snd_pcm arc4 intel_rapl x86_pkg_temp_thermal intel_powerclamp coretemp snd_seq_midi snd_seq_midi_event kvm_intel cmac snd_rawmidi kvm irqbypass crct10dif_pclmul i915 snd_seq crc32_pclmul drm_kms_helper aesni_intel rfcomm bnep aes_x86_64 iwlmvm lrw ir_xmp_decoder gf128mul mac80211 glue_helper ablk_helper cryptd ir_lirc_codec lirc_dev usbhid ir_mce_kbd_decoder ir_sony_decoder drm snd_seq_device ir_sanyo_decoder ir_sharp_decoder snd_timer i2c_algo_bit ir_jvc_decoder iwlwifi fb_sys_fops syscopyarea sysfillrect btusb ir_rc6_decoder hci_uart btrtl btbcm btqca sysimgblt btintel cfg80211 ir_rc5_decoder mei_me bluetooth ir_nec_decoder idma64 snd virt_dma mei soundcore shpchp intel_lpss_pci rc_rc6_mce binfmt_misc ite_cir rc_core video i2c_hid pinctrl_sunrisepoint hid pinctrl_intel intel_lpss_acpi intel_lpss nls_iso8859_1 acpi_pad acpi_als parport_pc kfifo_buf mac_hid industrialio ppdev lp parport e1000e ptp pps_core sdhci_pci sdhci ahci libahci fjes [last unloaded: videobuf2_memops]
[ 4610.982758] CPU: 3 PID: 5039 Comm: kworker/u8:2 Tainted: G        W  OE   4.4.0-040400-generic #201601101930
[ 4610.982760] Hardware name:                  /NUC6i5SYB, BIOS SYSKLi35.86A.0042.2016.0409.1246 04/09/2016
[ 4610.982775] Workqueue: i915-hangcheck i915_hangcheck_elapsed [i915]
[ 4610.982778]  0000000000000000 0000000079494344 ffff88001d67b9a8 ffffffff813c8d94
[ 4610.982782]  ffff88001d67b9f0 ffff88001d67b9e0 ffffffff8107dba2 ffff88048bcc0f00
[ 4610.982785]  ffff88048a452228 000000000000007e ffff8803b023fc00 ffff88048a452278
[ 4610.982789] Call Trace:
[ 4610.982795]  [<ffffffff813c8d94>] dump_stack+0x44/0x60
[ 4610.982799]  [<ffffffff8107dba2>] warn_slowpath_common+0x82/0xc0
[ 4610.982803]  [<ffffffff8107dc3c>] warn_slowpath_fmt+0x5c/0x80
[ 4610.982808]  [<ffffffff810e87b1>] ? internal_add_timer+0x71/0x80
[ 4610.982832]  [<ffffffffc06f151b>] intel_logical_ring_begin+0x1ab/0x250 [i915]
[ 4610.982854]  [<ffffffffc06f46e3>] emit_mocs_control_table+0x33/0x190 [i915]
[ 4610.982858]  [<ffffffff81095083>] ? try_to_grab_pending+0xb3/0x160
[ 4610.982880]  [<ffffffffc06f48fc>] intel_rcs_context_init_mocs+0xbc/0x330 [i915]
[ 4610.982901]  [<ffffffffc06f1f52>] gen8_init_rcs_context+0x182/0x290 [i915]
[ 4610.982921]  [<ffffffffc06f2085>] ? intel_logical_ring_reserve_space+0x25/0x30 [i915]
[ 4610.982939]  [<ffffffffc06d0ac8>] i915_gem_context_enable+0x28/0x50 [i915]
[ 4610.982960]  [<ffffffffc06e3799>] i915_gem_init_hw+0x1a9/0x4f0 [i915]
[ 4610.982972]  [<ffffffffc06a6cb0>] i915_reset+0x80/0x160 [i915]
[ 4610.982984]  [<ffffffffc06ab10a>] i915_reset_and_wakeup+0xea/0x170 [i915]
[ 4610.982997]  [<ffffffffc06afd1e>] i915_handle_error+0xce/0x630 [i915]
[ 4610.983002]  [<ffffffff810d4a0e>] ? vprintk_emit+0x2de/0x530
[ 4610.983006]  [<ffffffff810d4de9>] ? vprintk_default+0x29/0x40
[ 4610.983019]  [<ffffffffc06b0535>] i915_hangcheck_elapsed+0x275/0x470 [i915]
[ 4610.983023]  [<ffffffff8109688a>] process_one_work+0x1aa/0x440
[ 4610.983027]  [<ffffffff81096b6b>] worker_thread+0x4b/0x4c0
[ 4610.983030]  [<ffffffff81096b20>] ? process_one_work+0x440/0x440
[ 4610.983036]  [<ffffffff8109cce8>] kthread+0xd8/0xf0
[ 4610.983040]  [<ffffffff8109cc10>] ? kthread_create_on_node+0x1a0/0x1a0
[ 4610.983045]  [<ffffffff817fdd4f>] ret_from_fork+0x3f/0x70
[ 4610.983049]  [<ffffffff8109cc10>] ? kthread_create_on_node+0x1a0/0x1a0
[ 4610.983052] ---[ end trace 2256226d62c94ffa ]---

I looked online. Seem very little related to Realsesense and Skylake. But there are mentions that Skylake Linux graphic acceleration drivers have some bugs. The work around right now is to disable the HW acceleration. The drivers wont be fixed until kernel 4.6. However, I tested VLC which uses acceleration, it works fine, even at 4K.

Can someone look into this?

ddiakopoulos, 2016-07-31:
Unless I am mistaken, this issue isn't related to librealsense or RealSense per-say, but something having to do with the graphics driver on your SKL system when it opens a GL context. Unfortunately I don't think there's any actionable details on this issue for us.

closed 2016-07-31 by ddiakopoulos

robertxchen66, 2016-08-01:
I just turned of RC6 and it works fine now. So is there conflicts with RC6? The OS turns on RC6 by default with SKL. I have not test older processor with RC6 on. Just like to know if they are related.