realsenseai/librealsense #224: Video4Linux backend patching

tquang, 2016-07-29:
| Required Info  |   |
|---|---|
| Camera Model | R200 and SR300 | 
| Firmware Version | K501ux.209  | 
| Operating System & Version |  Ubuntu 16.04.1 LTS|
| Kernel Version (Linux Only)  | 4.2.0-3-upboard  |
| Build System |  Makefile | 

Hi, 
I have problems with detection of the built-in webcam, the real-sense and other cameras after running $ ./scripts/patch-uvcvideo-16.04.simple.sh. I got into many questions for "restart config.." while running the patching. I ignored by just entering all settings with nothing. Eventually I got 
"Unloading existing uvcvideo driver...
Script has completed. Please consult the installation guide for further instruction."
and done "sudo modprobe uvcvideo". Then, I got errors:
$ sudo dmesg | tail -n 50
[ 6055.565439]  [<ffffffff810b0391>] ? put_prev_entity+0x31/0x4b0
[ 6055.565454]  [<ffffffff8101efc9>] ? sched_clock+0x9/0x10
[ 6055.565471]  [<ffffffff81092b63>] process_one_work+0x153/0x400
[ 6055.565485]  [<ffffffff810932ee>] worker_thread+0x11e/0x480
[ 6055.565503]  [<ffffffff810931d0>] ? rescuer_thread+0x310/0x310
[ 6055.565516]  [<ffffffff81098a4b>] kthread+0xcb/0xf0
[ 6055.565529]  [<ffffffff81098980>] ? kthread_create_on_node+0x1c0/0x1c0
[ 6055.565544]  [<ffffffff817e8a1f>] ret_from_fork+0x3f/0x70
[ 6055.565557]  [<ffffffff81098980>] ? kthread_create_on_node+0x1c0/0x1c0
[ 6055.565569] ---[ end trace 8dceb5d805a8f1a9 ]---
After that, I can not run cheese to open my webcam and "cannot access '/dev/video*': No such file or directory".

l3s777, 2016-08-08:
I followed the installation guide, it worked well, I turned off the PC (linux running) and the next day... I got this error:

RealSense error calling rs_create_context(api_version:4):
    Cannot access /sys/class/video4linux

Now... should I install everything again?

Thanks!

tquang, 2016-08-08:
I reinstalled Linux, now it works. I'm able to run some given examples. But, only the color camera is working (attached is the result when I run cpp_alignimages). I also could not open the IR camera using cheese. l3s777, did you have the same result when you run cpp_alignimages? Were you able to see the IR video? 
Any help would be appreciated!

tquang, 2016-08-08:
I reinstalled Linux, now it works. I'm able to run some given examples. But, only the color camera is working (attached is the result when I run cpp_alignimages). I also could not open the IR camera using cheese. l3s777, did you have the same result when you run cpp_alignimages? Were you able to see the IR video? 
Any help would be appreciated!
[image]

l3s777, 2016-08-08:
hi ! when trying "cpp_alignimages" it works fine! the problem I have is when I turned off the computer and later turn on and it seems that librealsense is not installed :/

tquang, 2016-08-09:
The cameras are working now after I followed the instructions in https://github.com/IntelRealSense/librealsense/issues/70 for Video4Linux backend patching.

closed 2016-08-09 by tquang