realsenseai/librealsense #226: F200 camera locked at 15 FPS

dareversat, 2016-08-01:
Hello,

I am using an F200 camera on Ubuntu 14.04, kernel:  3.18.7-rt2-preempt
F200 firmaware:  2.60.0.0

I tried a little code on Qt: 

```
rs::context m_realSenseContext;
rs::device * m_f200;
if(m_realSenseContext.get_device_count() == 0)
        return;
m_f200 = m_realSenseContext.get_device(0);
m_f200->set_option(rs::option::f200_laser_power,15);
m_f200->set_option(rs::option::f200_accuracy,1);
m_f200->set_option(rs::option::f200_motion_range,50);
m_f200->set_option(rs::option::f200_filter_option,5);
m_f200->set_option(rs::option::f200_confidence_threshold,0);
m_f200->enable_stream(rs::stream::depth, 640, 480, rs::format::z16, 60);
while(true)
{
    m_f200->wait_for_frames();
    std::cout << m_f200->get_frame_timestamp(rs::stream::depth) << std::endl;
}
```
And there is the output:

```
0
69
138
207
277
346
415
485
554
623
693
762
831
900
970
1039
1108
1178
1247
1316
1386
1455
1524
1594
1663
1732
1801
1871
1940
2009
2079
2148
2217
2287
2356
2425
2495
2564
2633
2702
2772
2841
2910
2980
3049
3118
3188
3257
3326
3395
3465
3534
3603
3673
3742
3811
3881
3950
4019
4089
4158
4227
4296
4366
4435
4504
4574
4643
4712
4782
4851
```
I've got a frame each 70ms, but I set 60 FPS when I enable the stream, what am I missing?

Subquestions:
If I stop the program and start again, Linux crashes unless I plug in/out the camera before to restart, is there a log I can find to know what is going down?
At startup the camera is not recognized, I added the rules in rules.d but I have to plug out and plug in the camera at each startup, what is wrong?

dorodnic, 2016-08-01:
On the F200 / SR300 rs::option::f200_motion_range loosely translates to exposure time of the IR sensor. By setting f200_motion_range to 50, you are requesting exposure time longer then 16ms allowed by 60FPS, effectively reducing the frame-rate. By setting high f200_motion_range value you are prioritizing longer range over sensitivity to motion.

dareversat, 2016-08-01:
I understand, so this camera is not a 60 FPS camera, because when I set the exposure time to 0 I should have 60 FPS but I only get 30. Does the enable_stream function opens really the video stream at 60 FPS ?

dorodnic, 2016-08-01:
The stream is configured at 60 FPS. If you will run the same test with f200_motion_range = 0 you will get 60. However, setting f200_motion_range to 50 means depth exposure >16ms effectively reducing the actual FPS.

dareversat, 2016-08-01:
I clearly understood the problem, but with this code:
```
rs::context m_realSenseContext;
rs::device * m_f200;
if(m_realSenseContext.get_device_count() == 0)
        return;
m_f200 = m_realSenseContext.get_device(0);
m_f200->set_option(rs::option::f200_laser_power,15);
m_f200->set_option(rs::option::f200_accuracy,1);
m_f200->set_option(rs::option::f200_motion_range,0);
m_f200->set_option(rs::option::f200_filter_option,5);
m_f200->set_option(rs::option::f200_confidence_threshold,0);
m_f200->enable_stream(rs::stream::depth, 640, 480, rs::format::z16, 60);
while(true)
{
    m_f200->wait_for_frames();
    std::cout << m_f200->get_frame_timestamp(rs::stream::depth) << std::endl;
}
```

I've got this output:

```
0
33
66
99
133
166
199
232
266
299
332
365
399
432
465
498
532
565
598
632
665
698
731
765
798
831
864
898
931
964
997
1031
1064
1097
1130
1164
1197
1230
1264
1297
1330
1363
1397
1430
1463
1496
1530
1563
1596
1629
1663
1696
1729
1762
1796
1829
1862
1896
1929
1962
1995
```

So that make me think that this camera doesn't support 60 FPS on linux, but only 30 FPS.

dorodnic, 2016-08-01:
Please run the following snippet:
```
    rs::context m_realSenseContext;
    rs::device * m_f200;
    m_f200 = m_realSenseContext.get_device(0);
    m_f200->set_option(rs::option::f200_laser_power,15);
    m_f200->set_option(rs::option::f200_accuracy,3);
    m_f200->set_option(rs::option::f200_motion_range,0);
    m_f200->set_option(rs::option::f200_filter_option,5);
    m_f200->set_option(rs::option::f200_confidence_threshold,0);
    m_f200->enable_stream(rs::stream::depth, 640, 480, rs::format::z16, 60);
    m_f200->start();
    while(true)
    {
        m_f200->wait_for_frames();
        std::cout << m_f200->get_frame_timestamp(rs::stream::depth) << std::endl;
    }

```
I'm getting stable 60 fps on Linux:
```
0
16
33
49
66
83
99
116
```
Are you trying to optimize the various controls for some specific usage? With laser_power at 15 and motion_vs_range at 0 you are still getting good depth stream at 60 fps.

dareversat, 2016-08-02:
With these parameters I still got:
```
0
33
66
99
133
```
Yes those parameters are tested on a robot, but with motion range at 0, I've got issues with the depth when the robot is moving, there are holes in surfaces and that's why I set a higher motion range.
I think there is a problem with my usb controller or the preempt kernel I am using. If so, I can't change those two so I will work with 30 fps.

closed 2016-08-03 by dorodnic