realsenseai/librealsense #227: sample apps cannot run

robertxchen66, 2016-08-02:
| Required Info  |   |
|---|---|
| Camera Model | R200 | 
| Firmware Version | 1.0.72.06 | 
| Operating System & Version |14.04.5  |
| Kernel Version (Linux Only)  |  4.4.0 |
| Build System |   Makefile | 

I followed steps to update and patch the uvcvideo driver. After that, when I ran the examples like cpp-cature, there is a window frame pop up, but no video. The message on terminal is this:

desktop:~/librealsense$ ./bin/cpp-capture 
Capturing DEPTH at 480 x 360, fov = 56.0 x 43.5, distortion = NONE
Capturing COLOR at 640 x 480, fov = 54.5 x 41.9, distortion = MODIFIED_BROWN_CONRADY
Capturing INFRARED at 480 x 360, fov = 56.0 x 43.5, distortion = NONE
Capturing INFRARED2 at 480 x 360, fov = 56.0 x 43.5, distortion = NONE
libEGL warning: failed to create a pipe screen for i965

Any idea what's wrong?

xuyongzhi, 2016-08-02:
I am in trouble of this error too: " libEGL warning: failed to create a pipe screen for i965 ". I use SR300 camera.  Operating System  Version : 14.04.

vinjn, 2016-08-03:
First you can test `cpp-headless.cpp` to see if it's a bug from libEGL or librealsense.

robertxchen66, 2016-08-03:
Thanks. Here is the output of the cpp-headless:

desktop:~/librealsense$ ./bin/cpp-headless 
There are 1 connected RealSense devices.

Using device 0, an Intel RealSense R200
    Serial number: 2431008872
    Firmware version: 1.0.72.06
rs.warn: Subdevice 0 bad magic number 0x20000e
rs.warn: Subdevice 0 bad magic number 0xd000b
rs.warn: Subdevice 0 bad magic number 0xc000b
rs.warn: Subdevice 0 bad magic number 0xb000a
rs.warn: Subdevice 0 bad magic number 0x1a001a
rs.warn: Subdevice 0 bad magic number 0x1f001e
rs.warn: Subdevice 0 bad magic number 0x18001f
rs.warn: Subdevice 0 bad magic number 0x1c001a
rs.warn: Subdevice 0 bad magic number 0x1c0017
rs.warn: Subdevice 0 bad magic number 0x180018
rs.warn: Subdevice 0 bad magic number 0x1c001e
rs.warn: Subdevice 0 bad magic number 0x20001a
rs.warn: Subdevice 0 bad magic number 0x160019
rs.warn: Subdevice 0 bad magic number 0x1b001b
rs.warn: Subdevice 0 bad magic number 0x1c0020
rs.warn: Subdevice 0 bad magic number 0x1a001f
rs.warn: Subdevice 0 bad magic number 0x1a0018
rs.warn: Subdevice 0 bad magic number 0x18001d
rs.warn: Subdevice 0 bad magic number 0x190019
rs.warn: Subdevice 0 bad magic number 0x1c0021
rs.warn: Subdevice 0 bad magic number 0x1f0019
rs.warn: Subdevice 0 bad magic number 0x1a001a
rs.warn: Subdevice 0 bad magic number 0x1a001c
rs.warn: Subdevice 0 bad magic number 0x1c001b
rs.warn: Subdevice 0 bad magic number 0x1c0019
rs.warn: Subdevice 0 bad magic number 0x1e001c
rs.warn: Subdevice 0 bad magic number 0x1b001c
rs.warn: Subdevice 0 bad magic number 0x16001a
rs.warn: Subdevice 0 bad magic number 0x1a001e
rs.warn: Subdevice 0 bad magic number 0x1e001f
rs.warn: Subdevice 0 bad magic number 0x1b001c
rs.warn: Subdevice 0 bad magic number 0x18001b
rs.warn: Subdevice 0 bad magic number 0x1a001d
rs.warn: Subdevice 0 bad magic number 0x1f0025
rs.warn: Subdevice 0 bad magic number 0x180019
rs.warn: Subdevice 0 bad magic number 0x1e001b
rs.warn: Subdevice 0 bad magic number 0x1a0020
rs.warn: Subdevice 0 bad magic number 0x160018
rs.warn: Subdevice 0 bad magic number 0x160018
rs.warn: Subdevice 0 bad magic number 0x1c0019
rs.warn: Subdevice 0 bad magic number 0x1b0019
rs.warn: Subdevice 0 bad magic number 0x19001c
rs.warn: Subdevice 0 bad magic number 0x22001f
rs.warn: Subdevice 0 bad magic number 0x18001d
rs.warn: Subdevice 0 bad magic number 0x1a001c
rs.warn: Subdevice 0 bad magic number 0x19001e
rs.warn: Subdevice 0 bad magic number 0x1a0019
rs.warn: Subdevice 0 bad magic number 0x1c001c
rs.warn: Subdevice 0 bad magic number 0x1f001c
rs.warn: Subdevice 0 bad magic number 0x1b001e
rs.warn: Subdevice 0 bad magic number 0x1b001e
rs.warn: Subdevice 0 bad magic number 0x17001a
rs.warn: Subdevice 0 bad magic number 0x1d001c
rs.warn: Subdevice 0 bad magic number 0x1e001b
rs.warn: Subdevice 0 bad magic number 0x1c001d
rs.warn: Subdevice 0 bad magic number 0x19001d
rs.warn: Subdevice 0 bad magic number 0x1c001c
rs.warn: Subdevice 0 bad magic number 0x17001c
rs.warn: Subdevice 0 bad magic number 0x1c001c
rs.warn: Subdevice 0 bad magic number 0x1d001b
rs.warn: Subdevice 0 bad magic number 0x1d001e
rs.warn: Subdevice 0 bad magic number 0x1c0019
rs.warn: Subdevice 0 bad magic number 0x190019
rs.warn: Subdevice 0 bad magic number 0x1c001c
rs.warn: Subdevice 0 bad magic number 0x20001b
rs.warn: Subdevice 0 bad magic number 0x20001d
rs.warn: Subdevice 0 bad magic number 0x19001e
rs.warn: Subdevice 0 bad magic number 0x1a0018
rs.warn: Subdevice 0 bad magic number 0x1a001a
rs.warn: Subdevice 0 bad magic number 0x1a001d
rs.warn: Subdevice 0 bad magic number 0x1b001c
rs.warn: Subdevice 0 bad magic number 0x1c0019
rs.warn: Subdevice 0 bad magic number 0x1c0019
rs.warn: Subdevice 0 bad magic number 0x190019
rs.warn: Subdevice 0 bad magic number 0x21001e
rs.warn: Subdevice 0 bad magic number 0x1c001a
rs.warn: Subdevice 0 bad magic number 0x18001b
rs.warn: Subdevice 0 bad magic number 0x1d001a
rs.warn: Subdevice 0 bad magic number 0x1a001d
rs.warn: Subdevice 0 bad magic number 0x1c001c
rs.warn: Subdevice 0 bad magic number 0x1c001d
rs.warn: Subdevice 0 bad magic number 0x1a001d
rs.warn: Subdevice 0 bad magic number 0x1a001c
rs.warn: Subdevice 0 bad magic number 0x19001a
rs.warn: Subdevice 0 bad magic number 0x1b001c
rs.warn: Subdevice 0 bad magic number 0x19001d
rs.warn: Subdevice 0 bad magic number 0x19001a
wrote frames to current working directory.

closed 2016-11-19 by dorodnic