realsenseai/librealsense #228: Audio access on mac/linux

Rubikplayer, 2016-08-02:
Hi, I searched the library, and it seems that currently librealsense doesn't support audio access to realsense cameras (like in [windows SDK](https://software.intel.com/sites/landingpage/realsense/camera-sdk/v1.1/documentation/html/index.html?doc_speech_audio_data.html)). 

Q1: Is that correct? If so, is there any plan for supporting audio access in the future?

Actually, the purpose to access audio stream, is that I try to synchronize the realsense RGBD frames with an external video. Since the I  #cannot control that external camera programmatically, I can only sync them in post-processing. I thought about sync by matching the soundtracks. However the audio access to realsense seems not available for my setting. 

Q2: Does anyone has any idea for this sync problem? Any input will be helpful!

| Camera Model | SR300 | 
| Firmware Version | 3.15.0.0 | 
| Operating System & Version | OS X 10.11.5  |
| Build System | Makefile |

teknotus, 2016-08-02:
> Q2: Does anyone has any idea for this sync problem? Any input will be
helpful!

The SR300 audio is just a USB audio device. You can probably figure out
which sound device it is but it would be nice to have convenience functions
to match them up and record them together.

ddiakopoulos, 2016-08-05:
Hi @Rubikplayer - unfortunately audio support is not on our roadmap. Microphone integration only exists for some families of cameras (F200, SR300) and only in peripherals (there are many other embedded modules in the wild using librealsense). As @teknotus mentions, there is a separate USB audio interface that should be USB standards-compliant, so you should be able to access it using any cross-platform audio lib like RtAudio. Good luck, and let us know if you make any movement on the sync front using this technique!

closed 2016-08-05 by ddiakopoulos

Rubikplayer, 2016-08-06:
Thanks for your suggestions, @teknotus @ddiakopoulos !