realsenseai/librealsense #232: uvcvideo.ko crash after patching with patch-uvcvideo-16.04-simple.sh and modprobing

EricStrongSource, 2016-08-04:
| Required Info  |   |
|---|---|
| Camera Model | R200 | 
| Firmware Version | ? As-shipped from Intel  | 
| Operating System & Version | Linux Mint 18 Sarah  |
| Kernel Version (Linux Only)  | 4.4.0-21-generic  |
| Build System |  Makefile | 
`
[ 2108.799363] WARNING: CPU: 2 PID: 5092 at /build/linux-Ay7j_C/linux-4.4.0/drivers/media/v4l2-core/v4l2-ioctl.c:1272 v4l_enum_fmt+0xd86/0x10f0 [videodev]()
[ 2108.799365] Unknown pixelformat 0x30315752
[ 2108.799365] Modules linked in: uvcvideo(OE) videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_core v4l2_common videodev media xt_multiport iptable_filter ip_tables x_tables binfmt_misc drbg ansi_cprng eeepc_wmi intel_rapl snd_hda_codec_realtek asus_wmi dm_crypt x86_pkg_temp_thermal sparse_keymap mxm_wmi snd_hda_codec_generic intel_powerclamp snd_hda_intel coretemp snd_hda_codec kvm_intel snd_hda_core kvm snd_hwdep snd_pcm irqbypass snd_seq_midi crct10dif_pclmul crc32_pclmul snd_seq_midi_event snd_rawmidi aesni_intel snd_seq aes_x86_64 lrw gf128mul snd_seq_device glue_helper snd_timer ablk_helper cryptd snd serio_raw soundcore lpc_ich shpchp mei_me mei tpm_infineon wmi mac_hid acpi_pad sunrpc parport_pc ppdev lp parport autofs4 btrfs xor raid6_pq dm_mirror dm_region_hash dm_log i915 i2c_algo_bit
[ 2108.799392]  psmouse drm_kms_helper syscopyarea sysfillrect sysimgblt fb_sys_fops drm e1000e ptp video ahci libahci pps_core fjes [last unloaded: videobuf2_memops]
[ 2108.799398] CPU: 2 PID: 5092 Comm: AudioThread Tainted: G        W  OE   4.4.0-21-generic #37-Ubuntu
[ 2108.799400] Hardware name: ASUS All Series/H97I-PLUS, BIOS 2701 11/11/2015
[ 2108.799401]  0000000000000286 00000000ecd391a5 ffff8802147fbbf0 ffffffff813e93c3
[ 2108.799403]  ffff8802147fbc38 ffffffffc06e3a78 ffff8802147fbc28 ffffffff81080f62
[ 2108.799405]  ffff8802147fbda0 0000000000000000 0000000000000000 ffff88021153a900
[ 2108.799407] Call Trace:
[ 2108.799409]  [<ffffffff813e93c3>] dump_stack+0x63/0x90
[ 2108.799411]  [<ffffffff81080f62>] warn_slowpath_common+0x82/0xc0
[ 2108.799413]  [<ffffffff81080ffc>] warn_slowpath_fmt+0x5c/0x80
[ 2108.799416]  [<ffffffffc06fa72d>] ? uvc_ioctl_enum_fmt+0x9d/0xc0 [uvcvideo]
[ 2108.799420]  [<ffffffffc06ce6f6>] v4l_enum_fmt+0xd86/0x10f0 [videodev]
[ 2108.799426]  [<ffffffffc06cf2c1>] __video_do_ioctl+0x291/0x310 [videodev]
[ 2108.799428]  [<ffffffff81216c54>] ? terminate_walk+0x64/0xd0
[ 2108.799432]  [<ffffffffc06ced96>] video_usercopy+0x336/0x5b0 [videodev]
[ 2108.799437]  [<ffffffffc06cf030>] ? video_ioctl2+0x20/0x20 [videodev]
[ 2108.799442]  [<ffffffffc06cf025>] video_ioctl2+0x15/0x20 [videodev]
[ 2108.799446]  [<ffffffffc06ca6a3>] v4l2_ioctl+0xd3/0xe0 [videodev]
[ 2108.799448]  [<ffffffff812200ff>] do_vfs_ioctl+0x29f/0x490
[ 2108.799450]  [<ffffffff8121be34>] ? putname+0x54/0x60
[ 2108.799452]  [<ffffffff8120b85f>] ? do_sys_open+0x1bf/0x2a0
[ 2108.799453]  [<ffffffff81220369>] SyS_ioctl+0x79/0x90
[ 2108.799456]  [<ffffffff818244f2>] entry_SYSCALL_64_fastpath+0x16/0x71
[ 2108.799458] ---[ end trace 1b6c89db4cc7474a ]---
`

thielsn, 2016-08-18:
Hi,
I have the same issue, with f200 and a self-patched module for 4.7 kernel. Have you be able to solve the problem?

ev-mp, 2017-09-13:
@thielsn ,
The reported pixel format  0x30315752 is `RW10` , and is covered by LibRealSense kernel patch.
The patch itself has been refactored several times, including changing the underlying 4CC code for `RW10` so it is possible that the patched kernel module is not aligned with the formats definition that appear in source code.
In case it is still relevant you should reapply the [Linux backend installation guide](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md#video4linux-backend-preparation) and update accordingly.

closed 2017-09-13 by ev-mp