realsenseai/librealsense #234: rgb-depth alignment off with new SR300 package

adamwtow, 2016-08-05:
| Required Info  |   |
|---|---|
| Camera Model | SR300 | 
| Firmware Version | 3.15.0.0 | 
| Operating System & Version | Ubuntu 14.04.4 LTS |
| Kernel Version (Linux Only)  | 4.2.0-36-generic |
| Build System | Makefile | 

Hi,

We've got both the previous SR300 model and the new one that was recently released. The aligned streams in the new model are very poorly aligned when compared to the old model. Please see attached images. 

I've confirmed the same result when running the cameras through the Intel Realsense SDK on Win10 also. 

Wondering when this might be fixed and in the meantime, what is the best approach to performing our own intrinsic and extrinsic camera calibration for the SR300 on Ubuntu 14.04.

Example depth aligned to rgb stream image from the previous SR300 package:
[image]

Example depth aligned to rgb stream image from the recently released SR300 package:
[image]

l3s777, 2016-08-05:
hi! for this aligment are you using librealsense?

adamwtow, 2016-08-06:
Just using the stream: `rs::stream::depth_aligned_to_color`

Specifically using the following ROS wrapper a colleague of mine developed. No idea if it will work with the R200 though sorry. [https://github.com/jamessergeant/ros_realsense](url)

l3s777, 2016-08-08:
hi @adamwtow ! 
thanks for the reply... I have already tried:
uchar *depth_rgb = (uchar *) dev->get_frame_data(rs::stream::depth_aligned_to_color);
const uint8_t * rgb_depth_frame = reinterpret_cast<const uint8_t *>(depth_rgb);
cv::Mat rgb_depth_ = cv::Mat(480, 640, CV_8UC1, (void*)rgb_depth_frame);

later... plotting the last matrix in openCV (rgb_depth) which seems not to show the expected result.

ravenblackx, 2016-09-30:
I have a similar bad alignment using librealsense with an sr300 and code based on examples/cpp-pointcloud.cpp (I haven't tried with actual cpp-pointcloud because I'm running it on a headless server and providing the data to a separate client, but I'm *fairly* confident that the results would be the same with cpp-pointcloud.)

I've found the alignment is corrected if after calling
`rs_get_device_extrinsics(dev, RS_STREAM_DEPTH, RS_STREAM_COLOR, &extrin, &err);`
I multiply the three 'translation' values in extrin by 10000. I've tried this at different distances and the results are reliably good, vs. an error similar to that adamwtow depicted if I don't scale the translations up - if I took a frame of my hand with the fingers apart, the texture coordinates were consistently approximately one finger-and-gap width off from where they should have been, regardless of distance.

ravenblackx, 2016-10-03:
With library version 0.9.7, multiplying by 10000 seemed to align things correctly. I just updated to using library version 1.1.0 and now the error looks similar, but multiplying the translation values by 10000 now generates a [significantly] smaller error in the opposite direction. (With 0.9.7 the ceiling-colored glow around my fingers was small and evenly distributed on each side, but now it's all to one side.)

[image]

vs. without multiplying by 10000

[image]

(NB. this is from the 3D rendering in which any point without a depth doesn't get rendered at all.)

adamwtow, 2016-10-04:
I've pretty much fixed the issue now by running my own camera calibration for the intrinsics and extrinsics using a checkerboard and OpenCV and Matlab tools. In the process of creating an easy to use pipeline to calibrate new SR300 cameras really quickly but not sure when this will be finished yet. Haven't yet looked into calibrating depth for the cameras that have incorrect depth scale.

closed 2016-11-19 by dorodnic

severin-lemaignan, 2017-06-27:
@adamwtow I'm encountering the same issue. Could you share your intrinsics/extrinsics?

severin-lemaignan, 2017-06-28:
I still encounter the issue.

Two screenshots of `cpp-pointcloud`, same scene:

- the first one with the original SR300 package (RGB+D registration ok)

[image]

- the second one with the new SR300 package (noticeable offset of the RGB-D registration)

[image]

@dorodnic I think we should re-open this issue

adamwtow, 2017-06-29:
@severin-lemaignan sending you intrinsics/extrinsics won't help sorry as all the cameras are different. I think the best thing that can happen is Intel providing a calibration tool to make calibrating them easier. In the mean time feel free to trial some of the attached scripts (two cpp files and one python renamed to txt to allow upload). One is for capturing checkerboard images, one is for aligning depth image to 3x downsampled rgb image and the other is for calibrating based on checkerboard images. 

I hope the scripts are helpful but just forewarning that I cannot help at all with getting them to work. They are very messy and not commented.

[capture_checkerboard_images.txt](https://github.com/IntelRealSense/librealsense/files/1110722/capture_checkerboard_images.txt)

[align_depth_to_rgb.txt](https://github.com/IntelRealSense/librealsense/files/1110697/align_depth_to_rgb.txt)

[stereo_calibrate.txt](https://github.com/IntelRealSense/librealsense/files/1110712/stereo_calibrate.txt)

severin-lemaignan, 2017-06-29:
@adamwtow thanks for your feedback. I can certainly re-run an intrinsic/extrinsic calibration (I've done that many  times before), but I'm a bit puzzled to see such a difference between the two SR300 packages -- I believe Intel should provide updated calibration information for the new package...

kazoo-kmt, 2017-08-05:
@adamwtow, @severin-lemaignan, et al.
The above is helpful, but I still don't have concrete understanding. Could you confirm the followings about SR300? 
- Intrinsic parameters for both color camera and IR camera (=3D camera) are already calibrated by Intel when the SR300 is manufactured.
- It's not possible to change/modify these intrinsic parameters. (Or, is it possible to change by using SDK?)
- To calibrate the extrinsic parameters (R, T) for the color camera, we can use OpnCV's calibration method + checker board.
- Once after getting the extrinsic parameters of the color camera, we can calculate the extrinsic parameters for IR camera by using the transformation matrix between the color and the IR camera, which we can get by calling `get_extrinsics`.

honghande, 2019-01-07:
@ravenblackx，@adamwtow ，
Does this effect run Intel's SDK rendering? Can you tell me which one it is? Because Intel's color alignment of depth maps is not what we call the registration of depth maps and color maps? If I want to do the registration of depth maps and color maps, what about the demo in Intel?I look forward to your early reply. Thank you.