realsenseai/librealsense #237: Cannot access /sys/class/video4linux after completing installation and restarting the PC

l3s777, 2016-08-08:
| Required Info  |   |
|---|---|
| Camera Model | R200 | 
| Firmware Version |   | 
| Operating System & Version | Linux |
| Kernel Version (Linux Only)  |   |
| Build System | Makefile | 

I followed the installation guide, it worked well, I turned off the PC (linux running) and the next day... I got this error:

RealSense error calling rs_create_context(api_version:4):
Cannot access /sys/class/video4linux

I followed the installation guide, it worked well, I turned off the PC (linux running) and the next day... I got this error:

RealSense error calling rs_create_context(api_version:4):
Cannot access /sys/class/video4linux

Now... should I install everything again?

Thanks!

Now... should I install everything again?

Thanks!

ravrichard, 2016-08-09:
I am also experiencing a similar issue.

l3s777, 2016-08-09:
today I had to reinstall the stuff again and when I executed:
sudo dmesg | tail -n 50 (when Video4Linux backend)

I got this error: 
uvcvideo: module verification failed: signature and/or required key missing - tainting kernel

mgraugal, 2016-09-07:
I am also getting this error. I tried installing librealsense about four times, but this still appears. Anyone knows how to fix it?

l3s777, 2016-09-12:
not fixed yet! so everyday I need to reinstall uvc video :/

ev-mp, 2016-09-14:
You may need to modify the Makefile for the UVC driver to disregard driver signature verification:
**CONFIG_MODULE_SIG=n**

Here is a patch section that does exactly this (For Ubuntu 14.04LTS and 16.04 LTS)

diff --git a/drivers/media/usb/uvc/Makefile b/drivers/media/usb/uvc/Makefile
index c26d12f..d86cf22 100644
--- a/drivers/media/usb/uvc/Makefile
+++ b/drivers/media/usb/uvc/Makefile
@@ -1,3 +1,4 @@
+CONFIG_MODULE_SIG=n
 uvcvideo-objs  := uvc_driver.o uvc_queue.o uvc_v4l2.o uvc_video.o uvc_ctrl.o \
 		  uvc_status.o uvc_isight.o uvc_debugfs.o
 ifeq ($(CONFIG_MEDIA_CONTROLLER),y)

This section will later be incorporated into the pixel formats patch  to cope with more restrict drivers verification introduced in the latest Ubuntu updates

slomobile, 2016-09-19:
I also get this issue on my UP board robotics development kit.  Followed exact instructions here https://01.org/developerjourney/recipe/intel-realsense-robotic-development-kit#step1 Ubuntu 14.04LTS
Occurs following reboot with either F200 or R200.  Those same cameras work ok on a different windows 10 core i5 gen 3 machine.  Probable duplicate of closed issues #126, #132, and
https://software.intel.com/en-us/forums/realsense/topic/629246

user@box:~/librealsense$ bin/cpp-pointcloud
RealSense error calling rs_create_context(api_version:4):
    Cannot access /sys/class/video4linux

ScottJohnson2718, 2016-11-21:
I have the same error today with the UP board robotics development kit.  Any word on this?

ev-mp, 2017-10-08:
@l3s777 , @slomobile  do you still have issues with the driver not being reloaded after restart ?

closed 2017-10-17 by jlgarcia75