realsenseai/librealsense #238: uvcvideo cannot detect camera

jjpark7, 2016-08-09:
| Required Info  |   |
|---|---|
| Camera Model | R200| 
| Firmware Version |   | 
| Operating System & Version | Ubuntu 14.04  |
| Kernel Version (Linux Only)  | 4.4.0-040400-generic  |
| Build System |  Makefile | 

I can't run any of the examples provided
$ ./cpp-capture 
No device detected. Is it plugged in?

dmesg gives me
$ sudo dmesg |tail
[ 1624.195082] usb 2-1: SerialNumber: SN_0000000000
[ 1624.195850] uvcvideo: Found UVC 1.10 device Intel RealSense 3D Camera R200 (8086:0a80)
[ 1624.195957] uvcvideo: No streaming interface found for terminal 3.
[ 1624.196065] input: Intel RealSense 3D Camera R200 as /devices/pci0000:00/0000:00:14.0/usb2/2-1/2-1:1.0/input/input42
[ 1624.196577] uvcvideo: Found UVC 1.10 device Intel RealSense 3D Camera R200 (8086:0a80)
[ 1624.196677] uvcvideo: No streaming interface found for terminal 5.
[ 1624.196783] input: Intel RealSense 3D Camera R200 as /devices/pci0000:00/0000:00:14.0/usb2/2-1/2-1:1.2/input/input43
[ 1624.197253] uvcvideo: Found UVC 1.10 device Intel RealSense 3D Camera R200 (8086:0a80)
[ 1624.197398] uvcvideo: No streaming interface found for terminal 8.
[ 1624.197498] input: Intel RealSense 3D Camera R200 as /devices/pci0000:00/0000:00:14.0/usb2/2-1/2-1:1.4/input/input44

Also note,
$ lsusb
Bus 002 Device 009: ID 8086:0a80 Intel Corp. 
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 005: ID 8087:0a2b Intel Corp. 

So the device is correctly detected. 

When installing I already had gcc 4.9.3 and kernel 4.4.0 and installed openssl through apt-get.
Then, I ran ./scripts/patch-uvcvideo-4.4.sh v4.4-wily followed by sudo modprobe uvcvideo

Can anyone help me with this?

Thanks!

jjpark7, 2016-08-09:
I updated the firmware to 1.0.72.06 and now I can start ./cpp-capture. But now I only see the color stream not depth nor IR. Can anyone help me with this issue?

sudo dmesg | tail
[   24.739191] uvcvideo: Unknown video format 2036315a-0000-0010-8000-00aa00389b71
[   24.739194] uvcvideo: Found UVC 1.10 device Intel RealSense 3D Camera R200 (8086:0a80)
[   24.740681] uvcvideo: Unable to create debugfs 2-2 directory.
[   24.740777] input: Intel RealSense 3D Camera R200 as /devices/pci0000:00/0000:00:14.0/usb2/2-4/2-4:1.2/input/input24
[   24.742074] uvcvideo: Unknown video format 30315752-0000-0010-8000-00aa00389b71
[   24.742078] uvcvideo: Found UVC 1.10 device Intel RealSense 3D Camera R200 (8086:0a80)
[   24.743620] uvcvideo: Unable to create debugfs 2-2 directory.
[   24.743721] input: Intel RealSense 3D Camera R200 as /devices/pci0000:00/0000:00:14.0/usb2/2-4/2-4:1.4/input/input25
[   24.744636] usbcore: registered new interface driver uvcvideo
[   24.744637] USB Video Class driver (1.1.1)

jjpark7, 2016-08-09:
Reinstalled the whole librealsense after the firmware update solved all problems

closed 2016-08-09 by jjpark7

jacksonkr, 2016-11-07:
@jjpark7 How did you update the r200 firmware using ubuntu?

mgserafi, 2016-12-19:
I still get the error mentioned above. Where can the firmware be updated?

LDA169, 2017-02-12:
AFAIK, is not possible to update the firmware from ubuntu. Install Depth Camera Manager in Windows, the software will update the r200 firmware