realsenseai/librealsense #239: can't start camera and get uvc error when connect SRwith a usb3.0 hub

zhouxiangming18, 2016-08-10:
| Required Info  |   |
|---|---|
| Camera Model | SR300 | 
| Firmware Version |  3.1.25.2599 | 
| Operating System & Version | windows 10 x64  |
| Kernel Version (Linux Only)  |   |
| Build System |  VS2013 |

dorodnic, 2016-08-10:
Could you please attach what you see in device manager under imaging devices and under universal serial bus devices? In addition, please try to individually start color stream, depth stream and IR stream. Does color or depth by itself work?

zhouxiangming18, 2016-08-11:
@dorodnic thanks for your reply
[image]
[image]

zhouxiangming18, 2016-08-12:
@dorodnic is there something wrong?

jlgarcia75, 2017-10-03:
Hello,

If the issue is software related, we are no longer updating this software as we have moved to our latest generation, the D400 series. If you are in your warranty, a replacement can be provided.

Jesus G.
Intel Customer Support

closed 2017-10-03 by jlgarcia75