realsenseai/librealsense #241: get errorchange color_exposure frequently

zhouxiangming18, 2016-08-11:
| Required Info  |   |
|---|---|
| Camera Model | R200, F200 or SR300 | 
| Firmware Version |   | 
| Operating System & Version |   |
| Kernel Version (Linux Only)  |   |
| Build System |  VS2013, XCode or Makefile | 
camera->set_option(rs::option::color_exposure, 550)

dorodnic, 2016-08-11:
Please provide the error details. 
However, to the best of knowledge you will not be able to achieve this effect with color device due to hardware limitations. In addition, you should consider pipe-latency (even if the hardware would be able to set exposure in zero time, it would not necessarily affect the next frame you receive) 
SR300 Depth device does provide something similar using advanced settings, letting you set exposure for every n-th frame. This feature is not yet supported by librealsense.

zhouxiangming18, 2016-08-12:
@dorodnic  if I run at 30HZ,the result meet my requirements. but only lasted a few seconds, then I get the error: MF_E_VIDEO_RECORDING_DEVICE_INVALIDATED at camera->set_option(rs::option::color_exposure, 300)

jlgarcia75, 2017-10-17:
Hello @zhouxiangming18,

Do you still need help with this issue?

Regards,
Jesus G.
Intel Customer Support

closed 2017-10-26 by jlgarcia75