realsenseai/librealsense #243: Windows 10 Version 1607 causes camera to crash

Wollimayer, 2016-08-15:
| Required Info  |   |
|---|---|
| Camera Model | SR300 | 
| Firmware Version |  1.15.00 /  1.17.00   | 
| Operating System & Version |  Windows 10 : version 1607	os build14393.51  |
| Build System |  VS2015 | 

The librealses seems to have problems with the recent windows 10 Build ( version 1607 ) release on the 2.8.2016 .

I experince some random crashes of the camera. I could not pin done the exact line causing the issue but would recommand not to update to the current build yet.

(In case someone experinces the same problem, there is an option to revert the system back to the latest build ( without reinstalling the system)

Wollimayer, 2016-08-17:
SOLVED:

Problem was that 
ks_control->KsProperty((PKSPROPERTY)&node, sizeof(KSP_NODE), data, len, nullptr) does not accept nullpointer anymore. Changed to

ULONG bytes_received = 0;
            check("IKsControl::KsProperty", ks_control->KsProperty((PKSPROPERTY)&node, sizeof(KSP_NODE), data, len, &b

dorodnic, 2016-08-17:
Thank you, we'll make sure to integrate the fix

PR #245 (2016-08-17): Fixed set_control for Windows Build 1607
Fixed #243
KSProperty does not accept nullpointer in Windows Build 1607 (anniversary build)
closed 2016-08-17 by dorodnic