realsenseai/librealsense #250: Feature: Java Bindings

poqudrof, 2016-08-22:
Hello RealSense enthousiats, 

I developped a first Java binding for libRealsense: 

- The bindings are generated by [JavaCPP](https://github.com/bytedeco/javacpp). 
- They will be part of [JavaCPP-Presets](https://github.com/bytedeco/javacpp-presets), for now they are waiting in a pull request from my [fork](https://github.com/poqudrof/javacpp-presets/tree/realsense-pull). 
- They work with OpenCV in Java through JavaCV. 
- They will work with [Processing](https://processing.org/) and be part of the [PapARt library](https://project.inria.fr/papart/) . 

It is a first version, not a full feature yet. I can get color and depth, IR is also theoritically available.

If anyone want to contribute you are welcome to do so, here is what is missing: 
- Linux only. 
- Many Hacks in the source code to create the java bindings.
- Java overhead minimalistic (nothing really...).   

Cheers, 
Jeremy.

dorodnic, 2016-08-24:
Thank you, that's awesome. 
Can we link back to your fork for now from the documentation pages?

poqudrof, 2016-08-24:
Yes absolutely, it is not really easy to use for now, it will be conveniant once its integrated within javacpp-presets because precompiled binaries are distributed.

dorodnic, 2016-08-25:
Added a mention to Readme.md. Closing the issue for now, please update the community once it is integrated.

closed 2016-08-25 by dorodnic