realsenseai/librealsense #254: DKMS dependencies missing in ROS debian packages

tfoote, 2016-08-24:
Release 0.9.3 produces ROS debian packages that are uninstallable. @reganlo @mdhorn I've reverted the release into the rosdistros. https://github.com/ros/rosdistro/pull/12492 https://github.com/ros/rosdistro/issues/12493

Example of failing to install when setting up the realsense_camera package on top: http://build.ros.org/view/Ibin_uT64/job/Ibin_uT64__realsense_camera__ubuntu_trusty_amd64__binary/8/console

```
00:05:45.562 Setting up ros-indigo-librealsense (0.9.3-0trusty-20160823-153400-0700) ...
00:05:45.865 
00:05:45.865 Creating symlink /var/lib/dkms/uvcvideo/1.1.1-1-realsense/source ->
00:05:45.866                  /usr/src/uvcvideo-1.1.1-1-realsense
00:05:45.868 
00:05:45.869 DKMS: add completed.
00:05:45.986 Error! Your kernel headers for kernel 3.13.0-48-generic cannot be found.
00:05:45.987 Please install the linux-headers-3.13.0-48-generic package,
00:05:45.987 or use the --kernelsourcedir option to tell DKMS where it's located
00:05:45.991 libkmod: ERROR ../libkmod/libkmod.c:556 kmod_search_moddep: could not open moddep file '/lib/modules/3.13.0-48-generic/modules.dep.bin'
00:05:45.992 modinfo: ERROR: Module alias uvcvideo not found.
00:05:45.998 dpkg: error processing package ros-indigo-librealsense (--configure):
00:05:45.998  subprocess installed post-installation script returned error exit status 1
```

It looks like a missing dependency.

mdhorn, 2016-08-25:
@tfoote Thank you Tully.
Yes, in all the testing of DKMS I didn't account for the fact that the Docker images would be using the kernel of the Host OS instead of the Docker image.

I'll be adding the 'linux-headers-generic' as a dependency and testing install correctly under Docker. It looks like I'll have to add a few other dependencies for rebuilding kernel modules to the 'rosdistro' as well so that builds work under Docker.

Looking through the GitHub for the ROS build farm, I can't seem to find the specific Docker images that are used; any pointers?
I just want to ensure I test with the same Docker images before we resubmitted.

mdhorn, 2016-08-25:
As I suspected, even when the 'linux-headers-generic' is added as a dependency and are correctly installed, DKMS is searching for the kernel headers matching the Host OS kernel which are not installed:
```
Setting up linux-headers-generic (4.4.0.34.36) ...
Setting up ros-kinetic-librealsense (0.9.3-0xenial) ...
...
Creating symlink /var/lib/dkms/uvcvideo/1.1.1-1-realsense/source ->
                 /usr/src/uvcvideo-1.1.1-1-realsense
DKMS: add completed.
Error! Your kernel headers for kernel 3.13.0-92-generic cannot be found.
Please install the linux-headers-3.13.0-92-generic package,
```

In this case 4.4.0 is the right kernel for Xenial, but the host OS is Trusty and has a 3.13 kernel.
This will require some more investigation to find a solution in the Docker environment.

tfoote, 2016-08-25:
Yeah, that's an interesting problem. I don't know an immediate solution either. 

The base images all come from this snippet: https://github.com/ros-infrastructure/ros_buildfarm/blob/master/ros_buildfarm/templates/snippet/from_base_image.Dockerfile.em

They are either the stock upstream 'ubuntu' or 'debian' or on the alternative platforms you  can find the base images here: https://hub.docker.com/u/osrf/ like ubuntu_i386  

I believe they're all generated from this version now: https://github.com/osrf/ubuntu_32bit-docker/pull/1

mdhorn, 2016-11-01:
This was resolved by preventing DKMS in docker images.

closed 2016-11-01 by mdhorn

hinoue, 2018-11-06:
I am also seeing this error on 16.04 docker containers.  Is there a workaround?

KaleabTessera, 2019-02-19:
Same issue here. Any workaround?

developer-mayuan, 2020-04-10:
Same issue here.