realsenseai/librealsense #255: The delay of R300

xuyongzhi, 2016-08-25:
I use R300 to run a positioning system. I receive position produced by R300 and Optitrack at the same time. I use highly dynamic moving curves to align and calculate the delay between R300 and Optitrack. I find the delay of R300 relative to Optitrack could be 100~150 ms. 
Using the same system, I changed to another camara. It's only 20 ms. So I am sure the 100 ms delay does not come from my position program. I use ARUCO which runs fast.

Why is the image collect delay is so big? 

Does Realsense does some calculation before output the image? For instance, software focus.

dorodnic, 2016-08-25:
Hi, 

There is no significant software processing steps like you mentioned. 
The pipeline for generating depth is considerably longer then usual the one on a web-cam. It still should not reach the values you'r reporting. 
150ms at 30 FPS means 5 frames are being queued somewhere:
* the camera itself does not have the memory to hold them
* the uvc-driver and the relevant framework (Media Foundation on Windows and V4L on Linux) do maintain internal frame-queues. On a fast PC with low CPU consumption, these should be empty at most times. What is the configuration that you are using? Can you give it a try on a different setup.
* librealsense does minimal image conversions on the frames. If you are running on Intel architecture, this code is well optimized and tested to take less then 2 ms. 
* librealsense also maintains an internal frame-queue when you are working with wait_for_frames API. This function is trying to synchronize between color and depth streams, causing extra latency (usually less then 15 ms) On the development branch we provide frame callbacks that let you be the first one to know about a new frame. You can give that a try.

Could you please specify what exact camera model you are using (as displayed in the device manager)?
Also, what kind of streaming do you enable? (color, depth, color+depth) At what resolution and FPS?

xuyongzhi, 2016-08-25:
| Required Info  |   |
|---|---|
| Camera Model |  SR300 | 
| Firmware Version | 3.15.0.0  | 
| Operating System & Version | Ubuntu 14.04  |
| Kernel Version (Linux Only)  | 4.4.0-040400-generic  |
| Build System |  Makefile | 

I use V4L currently. resolution: 640*480, fps=30HZ.

Voidminded, 2017-06-07:
Hi @xuyongzhi,

I have a quick question, how did you manage to use SR300 in the presence of MoCap system ?
I'm using OptiTrack __Prime 13W__ that emit  __850nm__ and using them with __SR300__ which's projector emits __860nm__ IR pattern has a huge interference and makes it impossible to use. 
Using __R200__ will suppress the the interference a lot (thanks to it's stereo IR), however I need the minimum range that SR300 offers.

closed 2017-10-17 by jlgarcia75