realsenseai/librealsense #257: Problem with 320x240@60 FPS

r4ghu, 2016-08-30:
| Required Info  |   |
|---|---|
| Camera Model | R200, F200 or SR300 | 
| Firmware Version | 3.15.0.0  | 
| Operating System & Version |  Ubuntu 15.04 |
| Kernel Version (Linux Only)  | 3.19.0-66-generic  |
| Build System | Qtcreator | 

I am currently working with SR300 model and I am facing problem with getting depth and color stream at 320x240@60 fps. I get the following error

terminate called after throwing an instance of 'rs::error'
what():  uvc subdevice 1 cannot provide 320x240:Z16@60Hz DEPTH

I don't find any problem getting 640x480@60fps but I didn't understand why I am not getting it work at 320x240 frame size.

ev-mp, 2016-09-05:
Both F200 and SR300 do not support 320X240 for depth stream.
You can choose between 640X480 or 640X240 modes only.

dorodnic, 2016-09-05:
Hi,
Please check out [supported_video_formats](https://github.com/IntelRealSense/librealsense/blob/development/doc/supported_video_formats.pdf) for up-to-date list of supported formats.

closed 2016-09-05 by dorodnic