realsenseai/librealsense #261: Erroe: Timeout waiting for frames when using Cmake

syedharoonalam, 2016-09-02:
| Camera Model | R200 |
| Operating System & Version |  Ubuntu 14.04 |
| Kernel Version (Linux Only)  | 4.4 |
| Build System |  Makefile | 

I am trying to use example file: cpp-tutorial-3-pointcloud.cpp using CMakeLists.txt (from examples folder) and building the project using Cmake 3.6.

The project compiles correctly using 'make' in terminal but crashes at runtime with error: **rs::error was thrown when calling rs_wait_for_frames(device:0x1f643f8): Timeout waiting for frames.**

There is no error when building .pro from librealsense-master/librealsense.qtcreator.

Can you please suggest if there anything else need to add in the cmake file ?

mdhorn, 2016-09-02:
@syedharoonalam The CMakeLists.txt file in librealsense was added to enable build librealsense for  the Robotic Operating System (ROS) using catkin, and wasn't intended for direct CMake use.
We are in the process of updating this to be standard CMake plus needed support for ROS.
Hope to have this update by the end of next week.

syedharoonalam, 2016-09-05:
Thanks @mdhorn. Please close the issue when its updated, so I will also get a notification.

dorodnic, 2016-10-03:
With help from @mdhorn, we just tagged a release with working CMake support. [Installation instructions](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md) have been updated and everything should work out-of-the box (or at least as good as the hand-crafted makefile on Ubuntu 14.04)
Closing the issue for now, feel free to re-open if the problem persists.

closed 2016-10-03 by dorodnic