realsenseai/librealsense #262: Depth stream: 640x480: ok 320x240: Timeout waiting for frames.

maciekd, 2016-09-04:
| Required Info  |   |
|---|---|
| Camera Model | R200 | 
| Firmware Version |  1.0.72.06 | 
| Operating System & Version | Ubuntu 16.04.1 LTS |
| Kernel Version (Linux Only)  | 4.4.0-040400-generic  |
| Build System |  Makefile | 

Hi,

I'm trying to get a low resolution (320x240) depth stream going on Ubuntu. 640x480 works fine but 320x240 repors an error:
`rs_wait_for_frames(): Timeout waiting for frames.`

I've tried various USB ports, with or without USB hub. Nothing seems to help.

I've tried the same camera on an OSX machine and it worked ok with 320x240.

Any help would be greatly appreciated, thanks.

maciekd, 2016-09-07:
No one?  I'm really struggling with this.

Is this something proven to work? Could it be a configuration problem on my Ubuntu machine?

dorodnic, 2016-09-07:
Just to clarify that we did try to reproduce your problem on Ubuntu 16.04 and the same firmware, unsuccessfully, so it is a tricky one. If you are getting depth at 640x480, this pretty much eliminates any setup problems with the UVC driver patch. 320x240 should definitely [work](https://github.com/IntelRealSense/librealsense/blob/master/doc/supported_video_formats.pdf).  
It is possible that someone else has encountered this problem and would be able to help.
Can you please publish the exact code snippet you are trying to run? We can help check that the code is correct.

maciekd, 2016-09-07:
A short code snippet is a bit tricky, I've created a JNA java-wrapper for librealsense. 

I'll start off by recompiling the c-tutorial-1-depth so it runs 320x240. Will get get back once I've tested this.

maciekd, 2016-09-07:
Btw, another issue I have is that the framerate setting does not seem to work. I can set 30 fps on my depth stream but wait_for_frames only blocks for around 16ms - giving me always 60 fps. Could this be related?

maciekd, 2016-09-08:
I've done some tests with c-tutorial-1-depth:

**640x480 and 628x468** 
Works fine.

**320x240, 332x252, 480x360, 492x372** 
Does not actually time out but the frame rate is extremely slow: around 1 frame per 3 seconds.

Any tips on what I could try? Reinstall patches? I'm running out of ideas here..

dorodnic, 2016-09-08:
Do you have a Windows machine to test this? (we need to eliminate hardware issue option)
In addition, I would suggest adjusting `cpp-callback-2` demo as well as it receives the frames directly, bypassing `wait_for_frames` extra logic.

maciekd, 2016-09-08:
I've tested the same hardware on OSX (see my first post). I tried cpp-callback-2 and it seems to work!

I'm using the c-api but I found a callback there too so I guess I'll try using that.

dorodnic, 2016-09-08:
I see. One benefit of Windows is that you can also compare to the official RealSense SDK, but you are right, the hardware is probably Ok. 
If the callback demo is working but streaming demos don't it means something is wrong with frame time-stamps. It would be interesting to see time-stamps log of the different streams. If you call `rs::log_to_file(rs::log_severity::debug, "log.txt")`; librealsense can produce such log for you. 
Let us know if callback C API is working out for you. We might need to add c-callback sample. Don't forget to call `rs_release_frame `on a frame handle when you are done with it.

maciekd, 2016-09-09:
Unfortunately no luck. I've tried:

**cpp-callback-2**
Seems to be working fine - the framerate setting works and all resolutions work

**c-tutorial-1-depth**
higer resolutions (around 640x480) work. Framerate setting does not work. Always streams at 60fps. Lower resolutions timeout or are very slow (1 frame per 3 seconds).

**my own code using callbacks (C-API)**
Same result as c-tutorial-1-depth

Tried logging while running c-tutorial-1-depth. All I got was a bunch this:
`WARN: Subdevice 1 bad magic number 0x0`

maciekd, 2016-09-09:
Ah. The c++ example uses all streams. The C stuff use only depth. When only running depth on the cpp-callback-2 example I get the same result.

So. Seems like only depth at low resolution does not work.

maciekd, 2016-09-09:
Ok more info. If I enable color stream everything works file (framerate settings, no timeouts).

I've got 7 cameras connected to my system and don't need the color-stream so I'd very much would like to have it disabled. :)

maciekd, 2016-09-09:
Hmm. Seems like I'm getting all blank frames in the lower resolutions. So I'm back to square one: only resolutions around 640x480 work. The difference is that when I have the color stream on I get frames and the correct framerate. But the frames are blank..

maciekd, 2016-09-10:
Well. Solved it, kind of. I started trying to reinstall everything in pure desperation and finally ended up in a state where the entire system would lock up (as in hard reboot) when connecting a camera.

After that I reinstalled the entire system with Ubuntu Server (I'll be running headless in production anyway), ran Ubuntu Desktop last time. Then I followed the latest installation instructions and now everything in working perfectly!

Sorry I can't be of more help.

dorodnic, 2016-09-11:
@maciekd Glad to hear it worked after all.

closed 2016-09-11 by dorodnic