realsenseai/librealsense #264: "terminate called after throwing an instance of 'std::runtime_error'   what():  VIDIOC_DQBUF error 19, No such device Aborted (core dumped)"

PaoloCorazza, 2016-09-06:
| Required Info  |   |
|---|---|
| Camera Model | SR300 | 
| Firmware Version | 3.17.0.0  | 
| Operating System & Version | Ubuntu 14.04.05 LTS  |
| Kernel Version (Linux Only)  | 4.4.0-36-generic |
| Build System | Makefile | 

> ```
./cpp-alignements 
root@ACER-62105987376:/home/soilsciences/Documents/librealsense-master/bin# ./c-tutorial-2-streams 
There are 1 connected RealSense devices.

Using device 0, an Intel RealSense SR300
    Serial number: 614205002887
    Firmware version: 3.17.0.0
terminate called after throwing an instance of 'std::runtime_error'
  what():  VIDIOC_DQBUF error 19, No such device
Aborted (core dumped)
```

> ```./cpp-enumerate-devices 
Device 0 - Intel RealSense SR300:
 Serial number: 614205002887
 Firmware version: 3.17.0.0
 USB Port ID: 2-1
 Supported options:
    COLOR_BACKLIGHT_COMPENSATION : 0 .. 4, 1
    COLOR_BRIGHTNESS : -64 .. 64, 1
    COLOR_CONTRAST : 0 .. 100, 1
    COLOR_EXPOSURE : 0 .. 0, 1
    COLOR_GAIN : 0 .. 128, 1
    COLOR_GAMMA : 100 .. 500, 1
    COLOR_HUE : -180 .. 180, 1
    COLOR_SATURATION : 0 .. 100, 1
    COLOR_SHARPNESS : 0 .. 100, 1
    COLOR_WHITE_BALANCE : 2800 .. 6500, 1
    COLOR_ENABLE_AUTO_EXPOSURE : 0 .. 0, 1
    COLOR_ENABLE_AUTO_WHITE_BALANCE : 0 .. 1, 1
    F200_LASER_POWER : 0 .. 16, 1
    F200_ACCURACY : 1 .. 3, 1
    F200_MOTION_RANGE : 0 .. 100, 1
    F200_FILTER_OPTION : 0 .. 7, 1
    F200_CONFIDENCE_THRESHOLD : 0 .. 15, 1
    SR300_DYNAMIC_FPS : 0 .. 0, 0
    SR300_AUTO_RANGE_ENABLE_MOTION_VERSUS_RANGE : 0 .. 2, 1
    SR300_AUTO_RANGE_ENABLE_LASER : 0 .. 1, 1
    SR300_AUTO_RANGE_MIN_MOTION_VERSUS_RANGE : -32768 .. 32767, 1
    SR300_AUTO_RANGE_MAX_MOTION_VERSUS_RANGE : -32768 .. 32767, 1
    SR300_AUTO_RANGE_START_MOTION_VERSUS_RANGE : -32768 .. 32767, 1
    SR300_AUTO_RANGE_MIN_LASER : -32768 .. 32767, 1
    SR300_AUTO_RANGE_MAX_LASER : -32768 .. 32767, 1
    SR300_AUTO_RANGE_START_LASER : -32768 .. 32767, 1
    SR300_AUTO_RANGE_UPPER_THRESHOLD : 0 .. 65535, 1
    SR300_AUTO_RANGE_LOWER_THRESHOLD : 0 .. 65535, 1
 Stream DEPTH - 9 modes:
  640	x 480	@ 60Hz	Z16	67.9 x 53.6 degrees
  640	x 480	@ 30Hz	Z16	67.9 x 53.6 degrees
  640	x 480	@ 15Hz	Z16	67.9 x 53.6 degrees
  640	x 480	@ 5Hz	Z16	67.9 x 53.6 degrees
  640	x 240	@ 110Hz	Z16	67.9 x 53.6 degrees
  640	x 240	@ 60Hz	Z16	67.9 x 53.6 degrees
  640	x 240	@ 30Hz	Z16	67.9 x 53.6 degrees
  640	x 240	@ 15Hz	Z16	67.9 x 53.6 degrees
  640	x 240	@ 5Hz	Z16	67.9 x 53.6 degrees
 Stream COLOR - 245 modes:
  1920	x 1080	@ 30Hz	YUYV	69.4 x 42.6 degrees
  1920	x 1080	@ 30Hz	RGB8	69.4 x 42.6 degrees
  1920	x 1080	@ 30Hz	BGR8	69.4 x 42.6 degrees
  1920	x 1080	@ 30Hz	RGBA8	69.4 x 42.6 degrees
[... 434 lines cut ...]
Starting Intel RealSense SR300... done.
terminate called after throwing an instance of 'std::runtime_error'
  what():  VIDIOC_DQBUF error 19, No such device
Aborted (core dumped)
```

./cpp-tutorial-1-depth 
There are 1 connected RealSense devices.

Using device 0, an Intel RealSense SR300
    Serial number: 614205002887
    Firmware version: 3.17.0.0
terminate called after throwing an instance of 'std::runtime_error'
  what():  VIDIOC_DQBUF error 19, No such device
Aborted (core dumped)

./c-tutorial-1-depth 
There are 1 connected RealSense devices.

Using device 0, an Intel RealSense SR300
    Serial number: 614205002887
    Firmware version: 3.17.0.0
terminate called after throwing an instance of 'std::runtime_error'
  what():  VIDIOC_DQBUF error 19, No such device
Aborted (core dumped)

dorodnic, 2016-09-07:
Hi, we have not encountered this issue, but a prime suspect is the USB controller. In particular if you are using an USB-hub the camera might be drawing too much power on start causing a reset. If you have several USB3 ports try switching between them. If you need a hub, consider using powered one. Another theory can be that one of the video interfaces is not enumerated. Please check that you see `video0` and `video1` under `/dev/` and in `/sys/class/video4linux`

PaoloCorazza, 2016-09-08:
The computer is branded new and the USB ports are both usb 3.1, so I wouldn't suspect a power issue. 
Already tried with both available USB, reinstalling Ubuntu and upgrading to 16.04 LTS but I'm having the same problem. 
I'm just able to stream with cpp-config-ui few seconds of RGB and then the script closes itself
```
~/src/librealsense/bin$ ./cpp-config-ui
terminate called after throwing an instance of 'std::runtime_error'
  what():  VIDIOC_DQBUF error 19, No such device
Aborted (core dumped)

```

By the way `video0` and `video1` are both existing, and under `/dev/` >` /sys/class/video4linux`, `video4linux` is existing as well.

PaoloCorazza, 2016-09-08:
The attached documents represents the complete output of dmesg
[Dmesg.tar.gz](https://github.com/IntelRealSense/librealsense/files/460915/Dmesg.tar.gz)

vadali, 2016-11-07:
your dmesg error log looks almost exactly like mine (#343)..

Any luck getting this to work?

RoboEvangelist, 2017-05-09:
I get this error when I try to run multiple cameras in parallel with callbacks.

When I run the cameras in a while loop (polling frames devices after device), then this does not happen. But in this way, if I want to, for example, save every new frame to a file .jpg file, then the FPS for all cameras drop to 20 FPS (from 30 FPS originally). The hardware/software is very delicate very delicate.

YaoQ, 2017-06-10:
Ubuntu 14.04, SR300 camera and Alienware R5 labtop, I got same error. Then I plugin the camera into the **powered USB3.0 HUB**, then it works.

rconnorlawson, 2017-07-12:
Environment Info | .
-- | --
Camera Model | SR300
Firmware Version | 3.15.0.0
Operating System & Version | Ubuntu 16.04.1 LTS
Kernel Version (Linux Only) | 4.8.0-58-generic
Computer | Alienware 17

Single camera works fine when directly connected via USB 3.0. This issue arises using the same camera on an Amazon Basics USB 3.0 hub powered by a 48W(12V/3A) adapter. No other USB devices are connected.

papr, 2017-07-19:
Duplicate to #418

saydulk, 2017-08-16:
root@block:~/bdsm/src# ./BDSM-FETISHd
terminate called after throwing an instance of 'std::runtime_error'
  what():  locale::facet::_S_create_c_locale name not valid
Aborted (core dumped)

ev-mp, 2017-08-20:
"VIDIOC_DQBUF error 19" -  camera disconnection from host event, in most cases occurs due to the insufficient power provided by host. 
There are several ways to mitigate the issue, or at least to its minimize its re-occurrence:
- Connect the depth camera via a self-powered USB3 hub.
- Do not connect additional USB devices (including Keyboard and Mouse) to the same USB hub/port
- Avoid using long (3m+) USB3 cables.
- On laptops - prefer working with the AC plugged-in.

Additional info can be found in threads, [#418](https://github.com/IntelRealSense/librealsense/issues/418), [#264](https://github.com/IntelRealSense/librealsense/issues/264),[#558](https://github.com/IntelRealSense/librealsense/issues/558), [487](https://github.com/IntelRealSense/librealsense/issues/487) and more (apply "error 19" filter on the "Issues" tab).

closed 2017-08-20 by ev-mp

RoboEvangelist, 2017-08-20:
I recommend using an Industrial USB 3.0 hub. That solved my problem.