realsenseai/librealsense #266: SR300 driver installation issue, distorted/flickering depth image and no infrared image

Uliv, 2016-09-06:
| Camera Model: SR300
| Firmware Version: How do I get the firmware version?
| Operating System & Version: Ubuntu 14.04.5 LTS
| Kernel Version (Linux Only): Linux UR5 4.4.0-34-generic #53~14.04.1-Ubuntu SMP Wed Jul 27 16:56:40 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux
| Build System: Makefile 

**Nvidia driver version:**

> cat /proc/driver/nvidia/version

> NVRM version: NVIDIA UNIX x86_64 Kernel Module  352.79  Wed Jan 13 16:17:53 PST 2016
> GCC version:  gcc version 4.8.4 (Ubuntu 4.8.4-2ubuntu1~14.04.3)

Hi,

I am trying to install the drivers for the SR300, but when I run `./bin/cpp-capture`, then the depth image flickers and is distorted. There is also no infrared image visible, see image below (it's a paper box on a sheet of paper):

[image]

**These are the steps and some results:**
`git clone https://github.com/IntelRealSense/librealsense.git`

**I followed the instructions:**
https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md
There I have chosen step: 
> (OR) Kernel in 14.04.xx

(I also tried section "Updated 4.4 Stable Kernel" in a different attempt, after restoring my system from a diskimage before attempting to install the realsense drivers)

**Some Interesting outputs, when running `./scripts/patch-uvcvideo-ubuntu-mainline.sh` :**

> gpgv: Signature made Fri 12 Aug 2016 07:41:55 AM EDT using RSA key ID CBEECEA3
> gpgv: Can't check signature: public key not found
> dpkg-source: warning: failed to verify signature on ./linux-lts-xenial_4.4.0-36.55~14.04.1.dsc

[....]

> patching file include/uapi/linux/videodev2.h
> Hunk #1 FAILED at 621.
> 1 out of 1 hunk FAILED -- saving rejects to file include/uapi/linux/videodev2.h.rej

**The driver uvcvideo seems to register fine:**

> sudo dmesg | tail -n 50:
> [ 1236.238623] media: Linux media interface: v0.10
> [ 1236.268272] Linux video capture interface: v2.00
> [ 1236.327816] usbcore: registered new interface driver uvcvideo
> [ 1236.327820] USB Video Class driver (1.1.1)
> 

**Running the following gives me 1 failed test:**

> ur5@UR5:~/Software/librealsense/bin/tests$ ./SR300-live-test 
> terminate called after throwing an instance of 'std::runtime_error'
>   what():  VIDIOC_DQBUF error 19, No such device
> 
> ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
> SR300-live-test is a Catch v1.2.1 host application.
> Run with -? for options
> 
> -------------------------------------------------------------------------------
> SR300 streams 640x240 depth
> -------------------------------------------------------------------------------
> unit-tests/unit-tests-live-sr300.cpp:162
> ...............................................................................
> 
> unit-tests/unit-tests-live-sr300.cpp:162: FAILED:
>   {Unknown expression after the reported line}
> due to a fatal error condition:
>   SIGABRT - Abort (abnormal termination) signal
> 
> ===============================================================================
> test cases:   5 |   4 passed | 1 failed
> assertions: 531 | 530 passed | 1 failed

dorodnic, 2016-09-10:
It seems the automatic uvcvideo patching has failed during `./scripts/patch-uvcvideo-ubuntu-mainline.sh`. Please give it another try (fetch another copy of librealsense and re-run `patch-uvcvideo-ubuntu-mainline.sh` there). The patch should apply without any issues. 
However, you can always take the [patch](https://github.com/IntelRealSense/librealsense/blob/master/scripts/realsense-camera-formats.patch) as a reference, manually modify the relevant source files, compile and replace uvcvideo.ko. That's exactly what `patch-uvcvideo-ubuntu-mainline.sh` supposed to do but it can give you a better understanding of where the problem is.

Uliv, 2016-09-12:
Hi Sergey,

Thanks. I will rerun the patch and also look at the patch itself. I hope that tells me something.

Uli

mattcorsaro1, 2016-09-12:
I'm having the same problem with `./scripts/patch-uvcvideo-ubuntu-mainline.sh`. I haven't fixed the `Hunk #1 FAILED at 621.` issue yet, but I believe I've fixed `gpgv: Can't check signature: public key not found` with:
```
gpg --keyserver keyserver.ubuntu.com --recv-keys CBEECEA3
gpg --no-default-keyring -a --export CBEECEA3 | gpg --no-default-keyring --keyring ~/.gnupg/trustedkeys.gpg --import -
```

mattcorsaro1, 2016-09-12:
The patch is incompatible with at least Linux 4.4.0-36 and 4.4.0-34.

The patch is trying to add definitions for `V4L2_PIX_FMT_Y8I`, `V4L2_PIX_FMT_Y12I`, and `V4L2_PIX_FMT_Z16` on lines [174](https://github.com/IntelRealSense/librealsense/blob/master/scripts/realsense-camera-formats.patch#L174), 175, and 176. However, these definitions already exist in my videodev2.h. `/usr/src/linux-headers-4.4.0-36/include/uapi/linux/videodev2.h` and `/usr/src/linux-headers-4.4.0-34/include/uapi/linux/videodev2.h` contain:
```
#define V4L2_PIX_FMT_JPGL	v4l2_fourcc('J', 'P', 'G', 'L') /* JPEG-Lite */
#define V4L2_PIX_FMT_SE401      v4l2_fourcc('S', '4', '0', '1') /* se401 janggu compressed rgb */
#define V4L2_PIX_FMT_S5C_UYVY_JPG v4l2_fourcc('S', '5', 'C', 'I') /* S5C73M3 interleaved UYVY/JPEG */
#define V4L2_PIX_FMT_Y8I      v4l2_fourcc('Y', '8', 'I', ' ') /* Greyscale 8-bit L/R interleaved */
#define V4L2_PIX_FMT_Y12I     v4l2_fourcc('Y', '1', '2', 'I') /* Greyscale 12-bit L/R interleaved */
#define V4L2_PIX_FMT_Z16      v4l2_fourcc('Z', '1', '6', ' ') /* Depth data 16-bit */

/* SDR formats - used only for Software Defined Radio devices */
#define V4L2_SDR_FMT_CU8          v4l2_fourcc('C', 'U', '0', '8') /* IQ u8 */
```

So to fix the issue, replace lines 170-176 of `scripts/realsense-camera-formats.patch` with:
```
@@ -624,6 +624,19 @@ struct v4l2_pix_format {
 #define V4L2_PIX_FMT_Y8I      v4l2_fourcc('Y', '8', 'I', ' ') /* Greyscale 8-bit L/R interleaved */
 #define V4L2_PIX_FMT_Y12I     v4l2_fourcc('Y', '1', '2', 'I') /* Greyscale 12-bit L/R interleaved */
 #define V4L2_PIX_FMT_Z16      v4l2_fourcc('Z', '1', '6', ' ') /* Depth data 16-bit */```

Uliv, 2016-09-13:
Matt and I were able to get the SR300 camera driver to install correctly.

closed 2016-09-13 by Uliv

changhai0551, 2016-09-28:
hello,
 I  encountered the same problem as yours, flickering depth image and no infrared image.I followed the the same instructions: https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md
my linux Kernel is:
Linux hai-ubuntu 4.4.0-040400-generic #201601101930 SMP Mon Jan 11 00:32:41 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux
 Operating System & Version: Ubuntu 14.04.5 LTS

I have read the above solutions, but I still can't fix my issue.

How can I handle my problem?
Thank you very much.

changhai0551, 2016-09-28:
I have installed the driver correctly.

changhai0551, 2016-09-28:
Just rerun the ./scripts/install_dependencies-4.4.sh and ./scripts/patch-uvcvideo-4.4.sh v4.4-wily and follow the next steps.

changhai0551, 2016-10-10:
Hi, Uliv 
 I encountered the same problem as yours, flickering depth image and no infrared image.
Also, sudo dmesg | tail -n 50: outputs
usbcore: registered new interface driver uvcvideo
could you tell me how you fix your problem?
Thanks you very much.