realsenseai/librealsense #267: Depth stream - Noisy at edges

r4ghu, 2016-09-06:
| Required Info  |   |
|---|---|
| Camera Model | SR300 | 
| Firmware Version | 3.15.0.0  | 
| Operating System & Version | Ubuntu 15.04  |
| Kernel Version (Linux Only)  | 3.19.0-66-generic  |
| Build System |  Qtcreator | 

I am trying to integrate the librealsense with [htrack](https://github.com/OpenGP/htrack). I wrote the following code to achieve the task - [Sensor_librealsense.cpp](https://github.com/OpenGP/htrack/blob/master/tracker/Sensor/Sensor_librealsense.cpp). By observing the file, you can see that I didn't do much of the processing for the depth stream. I only converted the raw depth value obtained from camera into depth (in meters).

I observed that the depth stream is noisy at the edges and this is interfering with further processing of our application. I tried applying median filter onto the depth stream (kernel sizes - 3 and 5) to remove this (salt and pepper) noise. But I didn't find any change (the kernel sizes of higher order can't be applied due to restriction of OpenCV library to [CV_16U](http://docs.opencv.org/2.4/modules/imgproc/doc/filtering.html?highlight=medianblur#medianblur) format). 

Video Link - https://youtu.be/yXLHDmjwYRw . 

How can we remove this unnecessary noise?

dorodnic, 2016-09-07:
As a general direction you might want to play around with various depth options. In particular different values for the `F200_FILTER_OPTION` will yield different visual results. `F200_CONFIDENCE_THRESHOLD` also has effect on how much noise can get in. 
In addition, you might want to check out the different algorithmic presets (`rs_ivcam_preset enum` in rs.h). Using `apply_ivcam_preset` you can configure the camera to predefined settings optimized for different types of usages.

r4ghu, 2016-09-07:
Issue solved! The problem is not with the camera but the post processing of the depth stream I was implementing. Since, the depth stream provided by the camera is 640x480 and the depth stream I am using is 320x240, I used the opencv's resize function. The problem with that function is that it does anti-aliasing while resizing the images.

closed 2016-09-07 by r4ghu