realsenseai/librealsense #270: No visual effects change with R200_DISPARITY options

wangji1, 2016-09-07:
| Required Info  |   |
|---|---|
| Camera Model | R200 | 
| Firmware Version | 1.0.72.06  | 
| Operating System & Version | Ubuntu 14.04 & 16.04 |
| Kernel Version (Linux Only)  | 4.4.0-040400-generic  |
| Build System | Makefile | 

From tag v1.9.7, in cpp-config-ui, I adjust the options R200_DISPARITY_MULTIPLIER and R200_DISPARITY_SHIFT, but not find any visual effects.
Is it normal? or I have something wrong?

dorodnic, 2016-09-07:
Please refer to comment in #271. Hope to come up with the answer soon

dorodnic, 2016-09-09:
Based on the best sources I could get, it seems that the options above are not implemented in the DS4 firmware. In the next release `is_supported` on these options should be changed to return false.

closed 2017-08-01 by oordan