realsenseai/librealsense #271: No visual effects change with R200_AUTO_EXPOSURE_KP options

wangji1, 2016-09-07:
| Required Info  |   |
|---|---|
| Camera Model | R200 | 
| Firmware Version | 1.0.72.06  | 
| Operating System & Version | Ubuntu 14.04 & 16.04  |
| Kernel Version (Linux Only)  | 4.4.0-040400-generic  |
| Build System | Makefile | 

From tag v1.9.7, since #149 has been fixed, I tried to validate R200_AUTO_EXPOSURE options in cpp-config-ui.
When adjust R200_AUTO_EXPOSURE_KP_EXPOSURE and R200_AUTO_EXPOSURE_KP_DARK_THRESHOLD, can't find any visual effects.
when adjust R200_AUTO_EXPOSURE_KP_GAIN, the depth and infrared streams will flicker.

Is it normal?

dorodnic, 2016-09-07:
I can confirm that these options have very limited (if any) visual effect and there is not much information on them in the R200 documents. We'll try to check with R200 developers what effect these are supposed to have. Please notice that on newer version of the R200 camera (LR200 and ZR300) these options has been removed.

dorodnic, 2016-09-08:
I can confirm that the options you mentioned are in fact deprecated and most likely don't have any visual effect. I will make sure they are removed from the list of options supported by R200 and eventually from librealsense API (at the next API breaking release).
General note about R200 **Auto-Exposure Region-of-Interest**:
This feature is native to LR200 and ZR300 cameras, but for the R200 it is provided as **experimental**. So from librealsense point of view, using R200_AUTO_EXPOSURE_ family of options should not crash the program or leak resources, however we can't promise good visual results.

closed 2016-09-08 by dorodnic