realsenseai/librealsense #273: depth/ir-images reloading in config-ui

sfdasalo, 2016-09-07:
| Camera Model | R200, F200 or SR300 | 
| Firmware Version |  3.10.10.0  | 
| Operating System & Version |  ubuntu 14.04 LTS |
| Kernel Version (Linux Only)  | 4.4.0-31-generic  |
| Build System |  Makefile | 

Hello there.
I've got a problem by using the implemented example cpp-config-ui.
It seems the streaming depth and ir images in the cpp-config-ui are rebooting or something like that in a delay of 3 seconds.
This problem only appears when I am using the parameter configuration 0 or 2 (common or longrange).
To show you exactly what i mean i show you the pictures below:
[image]
in this gif you can see how the image looks like it "cuts itself down" and reload again.

[image]
This picture shows you how the image streaming should be.

[image]
This picture shows you the direct "reload" of the camera.

I am always working with a warm camera, means I begin to shoot after five till ten minutes after camera using.

I hope the issue is nearly clear enough verbalised.

sfdasalo, 2016-09-15:
Okay - It seems it has nothing to do with the parameters. I recreated the issue with the IVCAM options preset 4 aswell. Someone an idea?

[image]

jlgarcia75, 2017-10-17:
Hello @sfdasalo ,

Do you still need help with this issue?

Regards,
Jesus G.
Intel Customer Support

closed 2017-10-26 by jlgarcia75