realsenseai/librealsense #275: QT Creator compile issue

RickBlacker, 2016-09-07:
| Required Info  |   |
|---|---|
| Camera Model | R200, F200 or SR300 | 
| Firmware Version |   | 
| Operating System & Version | Ubuntu  |
| Kernel Version (Linux Only)  |  16.04  |
| Build System |  VS2013, XCode or Makefile | 

All, I grabbed the latest master this morning. Loaded up the QT project in QT.  When trying to build the solution, the build is running but I'm getting non stop warnings that state :
file '../librealsense.qtcreateor/all.pro' has modification time 40168 s in the future.

This error is being duplicated over and over. In fact over 17,000 times now and there is no end in sight of the build process ending.

dorodnic, 2016-09-07:
In general this error occurs when you copy make file from computers with clocks that do not agree. In context of QT, I would recommend to right-click on the project and run qmake.

RickBlacker, 2016-09-07:
Ah.. ok

closed 2016-09-07 by dorodnic