realsenseai/librealsense #277: Trying to create a Eclipse project.

RickBlacker, 2016-09-07:
| Required Info  |   |
|---|---|
| Camera Model | R200, F200 or SR300 | 
| Firmware Version |   | 
| Operating System & Version | Ubuntu  |
| Kernel Version (Linux Only)  |  16.04 |
| Build System |  VS2013, XCode or Makefile | 

Hi all... I'm trying to create a project using Eclipse. I'm not having much luck. I've configured my project to point to the proper include directory, proper library directory, included the librealsense library. Yet, I'm getting LOTs of build errors in Eclipse. See pictures.   I was hoping someone might be able to help me isolate why I'd be getting these.
I've included various screen shots to help show you how I've configured as well as various errors I'm seeing and getting.  I have no doubt it's operator error. 

[image]
[image]
[image]
[image]
[image]

RickBlacker, 2016-09-07:
Update.... Got it figured out that in the Miscellaneous settings for GCC C++ Compiler. I had to set -std-c++11 manually under the "Other Flags" section.

RickBlacker, 2016-09-07:
So, getting past the build errors, I'm still having a linker issue. Even though the librealsense lib is in /usr/local/lib and I've set that path and the librealsense library in Eclipse, I'm getting a linker error saying it can't find the library.

Any ideas why?

RickBlacker, 2016-09-07:
Found out i'm supposed to use realsense not librealsense. Compiling and linking now.

dorodnic, 2016-09-08:
Hi @RickBlacker 
Hope you managed to figure out Eclipse issues, we currently do not provide out-of-the-box solution for Eclipse developers as we do with Qt. You are welcomed to publish the steps once you succeed, it can be useful to the community. 
Another way to support Eclipse developers would be through the use of CMake (we are still working on it)
Closing the issue for now, feel free to reopen and update on your findings.

closed 2016-09-08 by dorodnic