realsenseai/librealsense #280: videodev2.h fails patching

TheMikeyR, 2016-09-09:
| Required Info  |   |
|---|---|
| Camera Model | SR300 | 
| Firmware Version |  N/A | 
| Operating System & Version | Ubuntu 16.04  |
| Kernel Version (Linux Only)  |  4.4.0-36-generic |
| Build System |  Makefile | 

````
$ ./scripts/patch-uvcvideo-16.04.simple.sh
Reading package lists... Done
Building dependency tree       
Reading state information... Done
libusb-1.0-0-dev is already the newest version (2:1.0.20-1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Reading package lists... Done
Building dependency tree       
Reading state information... Done
build-essential is already the newest version (12.1ubuntu2).
linux-headers-generic is already the newest version (4.4.0.36.38).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
patching file drivers/media/usb/uvc/uvc_driver.c
Hunk #1 succeeded at 888 with fuzz 2 (offset 720 lines).
patching file drivers/media/usb/uvc/uvcvideo.h
Hunk #1 succeeded at 241 with fuzz 2 (offset 126 lines).
patching file include/uapi/linux/videodev2.h
Reversed (or previously applied) patch detected!  Assume -R? [n] 
Apply anyway? [n] y
Hunk #1 FAILED at 624.
1 out of 1 hunk FAILED -- saving rejects to file include/uapi/linux/videodev2.h.rej
```
videodev2.h.rej includes this:
```
--- include/uapi/linux/videodev2.h
+++ include/uapi/linux/videodev2.h
@@ -624,6 +624,19 @@ struct v4l2_pix_format {
 #define V4L2_PIX_FMT_Y8I      v4l2_fourcc('Y', '8', 'I', ' ') /* Greyscale 8-bit L/R interleaved */
 #define V4L2_PIX_FMT_Y12I     v4l2_fourcc('Y', '1', '2', 'I') /* Greyscale 12-bit L/R interleaved */
 #define V4L2_PIX_FMT_Z16      v4l2_fourcc('Z', '1', '6', ' ') /* Depth data 16-bit */
+#define V4L2_PIX_FMT_Y8       v4l2_fourcc('Y', '8', ' ', ' ') /* Greyscale 8-bit */
+#define V4L2_PIX_FMT_Y10      v4l2_fourcc('Y', '1', '0', ' ') /* Greyscale 10-bit */
+#define V4L2_PIX_FMT_Y12      v4l2_fourcc('Y', '1', '2', ' ') /* Greyscale 12-bit */
+#define V4L2_PIX_FMT_Y16      v4l2_fourcc('Y', '1', '6', ' ') /* Greyscale 16-bit */
+#define V4L2_PIX_FMT_RAW8     v4l2_fourcc('R', 'A', 'W', '8') /* Raw data 8-bit */
+#define V4L2_PIX_FMT_RW10     v4l2_fourcc('R', 'W', '1', '0') /* Raw data 10-bit */
+#define V4L2_PIX_FMT_RW16     v4l2_fourcc('R', 'W', '1', '6') /* Raw data 16-bit */
+#define V4L2_PIX_FMT_INVZ     v4l2_fourcc('I', 'N', 'V', 'Z') /* 16 Depth */
+#define V4L2_PIX_FMT_INZI     v4l2_fourcc('I', 'N', 'Z', 'I') /* 24 Depth/IR 16:8 */
+#define V4L2_PIX_FMT_INVR     v4l2_fourcc('I', 'N', 'V', 'R') /* 16 Depth */
+#define V4L2_PIX_FMT_INRI     v4l2_fourcc('I', 'N', 'R', 'I') /* 24 Depth/IR 16:8 */
+#define V4L2_PIX_FMT_INVI     v4l2_fourcc('I', 'N', 'V', 'I') /* 8 IR */
+#define V4L2_PIX_FMT_RELI     v4l2_fourcc('R', 'E', 'L', 'I') /* 8 IR alternating on off illumination */
 
 /* SDR formats - used only for Software Defined Radio devices */
 #define V4L2_SDR_FMT_CU8          v4l2_fourcc('C', 'U', '0', '8') /* IQ u8 */
```

Not sure exactly what goes on, I thought I managed to install it correct but when I ran the dmesg it said that the uvcvideo module verification failed. 

the camera wasn't plugged in during the entire installation any ideas what is wrong?
BTW it is a nearly fresh installed ubuntu 16.04

seer-it, 2017-05-03:
What was wrong ?

runjerry, 2017-07-13:
Hi Mike, I am having the same problem in patching videodev2.h and wonder if you could possibly share your solution to it? Thanks!

MikhailKazekin, 2017-07-17:
+1 to @runjerry, I can confirm I also see this and kindly ask @TheMikeyR to share the solution.

TheMikeyR, 2017-07-26:
@seer-it @runjerry @MikhailKazekin 
Sorry for being offline, I didn't manage to solve the solution, so I will reopen the issue.

My solution was to develop on Windows instead since I ran into multiple issues regarding linux and librealsense.

ev-mp, 2017-08-20:
Hi @TheMikeyR , Librealsense provides a [consolidated patch](https://github.com/IntelRealSense/librealsense/blob/master/scripts/patch-realsense-ubuntu-xenial.sh) for Ubuntu LTS16.04.1 and 16.04.2.
The patch has been verified to work correctly with kernels 4.4 and the recent 4.10.

Note that if you haven't updated Ubuntu recently, then it is preferable to rerun the relevant chapter from the [installation manual](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md#video4linux-backend-preparation) to avoid some common configuration/installation issues.

closed 2017-08-20 by ev-mp