realsenseai/librealsense #281: Video4Linux backend problems

henryleinhos, 2016-09-09:
| Required Info  |   |
|---|---|
| Camera Model | R200 | 
| Firmware Version |   1.0.72.06 | 
| Operating System & Version |  Ubuntu 14.04 LTS |
| Kernel Version (Linux Only)  |  4.4.0-040400-generic (x86_64) |
| Build System |  Makefile | 

Hi,

After upgrading my kernel on my embedded system with the included script:

```
 $ ./scripts/install_dependencies-4.4.sh
```
rebooting, and patching the new kernel module with:
```
 $ ./scripts/patch-uvcvideo-4.4.sh v4.4-wily 
```
 
I see the device show up in dmesg after plugging it in:

```
72.261623] usb 3-1: new SuperSpeed USB device number 2 using xhci_hcd
[   72.280675] usb 3-1: New USB device found, idVendor=8086, idProduct=0a80
[   72.280683] usb 3-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[   72.280687] usb 3-1: Product: Intel RealSense 3D Camera R200
[   72.280691] usb 3-1: Manufacturer: Intel Corp
[   72.280695] usb 3-1: SerialNumber: SN_2441004691
[   72.304251] media: Linux media interface: v0.10
[   72.318669] Linux video capture interface: v2.00
[   72.340169] uvcvideo: module verification failed: signature and/or required key missing - tainting kernel
[   72.343156] uvcvideo: Found UVC 1.10 device Intel RealSense 3D Camera R200 (8086:0a80)
[   72.346833] uvcvideo: Found UVC 1.10 device Intel RealSense 3D Camera R200 (8086:0a80)
[   72.348359] uvcvideo: Unable to create debugfs 3-2 directory.
[   72.348669] input: Intel RealSense 3D Camera R200 as /devices/pci0000:00/0000:00:14.0/usb3/3-1/3-1:1.2/input/input6
[   72.350107] uvcvideo: Found UVC 1.10 device Intel RealSense 3D Camera R200 (8086:0a80)
[   72.352310] uvcvideo: Unable to create debugfs 3-2 directory.
[   72.352513] input: Intel RealSense 3D Camera R200 as /devices/pci0000:00/0000:00:14.0/usb3/3-1/3-1:1.4/input/input7
[   72.353551] usbcore: registered new interface driver uvcvideo
[   72.353554] USB Video Class driver (1.1.1)
```

`lsusb` shows the camera ok:

```
$ lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 002: ID 8086:0a80 Intel Corp. 
Bus 003 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 002 Device 004: ID 8087:07dc Intel Corp. 
Bus 002 Device 003: ID 04b4:6560 Cypress Semiconductor Corp. CY7C65640 USB-2.0 "TetraHub"
Bus 002 Device 002: ID 0403:6015 Future Technology Devices International, Ltd Bridge(I2C/SPI/UART/FIFO)
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```
and I see 3 video devices in /dev (this unit is running headless):

```
$ ls /dev/video*
/dev/video0  /dev/video1  /dev/video2
```

When I run the demo code, however, I don't get any devices detected:

```
$ bin/cpp-enumerate-devices 
No device detected. Is it plugged in?
```

If I switch to the LIBUVC library backend, I can talk to the device (mostly) fine:

```
$ bin/cpp-enumerate-devices 
Device 0 - Intel RealSense R200:
 Serial number: 2441004691
 Firmware version: 1.0.72.06
 USB Port ID: 3-1
 Supported options:
    COLOR_BACKLIGHT_COMPENSATION : 0 .. 4, 1, 1
    COLOR_BRIGHTNESS : 0 .. 255, 1, 56
    COLOR_CONTRAST : 16 .. 64, 1, 32
    COLOR_EXPOSURE : 1 .. 666, 1, 1
    COLOR_GAIN : 0 .. 256, 1, 32
    COLOR_GAMMA : 100 .. 280, 1, 220
    COLOR_HUE : 0 .. 0, 32766, 3.19543e+08
    COLOR_SATURATION : 0 .. 255, 1, 128
    COLOR_SHARPNESS : 0 .. 7, 1, 0
    COLOR_WHITE_BALANCE : 2000 .. 8000, 1, 6500
    COLOR_ENABLE_AUTO_EXPOSURE : 0 .. 1, 1, 1
...
```

Not sure what's going on here...

henryleinhos, 2016-09-09:
Looking at this closer:  Is this a permissions issue?  I am running this as a regular user (unprivileged), and not a member of video group...

mdhorn, 2016-09-09:
@henryleinhos Yes, your account should be in the group of the /dev/video devices.
The assumption is your user account will be in default set of groups added by the UI;  sudo, video, plugdev, etc.

ev-mp, 2016-09-13:
@henryleinhos, please check that you applied correctly udev rules, as specified in [Linux Installation - Video4Linux Backend ](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md#video4linux-backend)  chapter ?

- sudo cp config/99-realsense-libusb.rules /etc/udev/rules.d/
- sudo udevadm control --reload-rules && udevadm trigger

closed 2016-11-19 by dorodnic