realsenseai/librealsense #282: RealSense error calling rs_create_context(api_version:10907):     api version mismatch

RickBlacker, 2016-09-09:
| Required Info  |   |
|---|---|
| Camera Model | F200 | 
| Firmware Version |   | 
| Operating System & Version |  Ubuntu 16.04 |
| Kernel Version (Linux Only)  |   |
| Build System |  Make file |

 All, I downloaded the latest master. Ran make, then sudo make install.  Everything builds fine but when I try to run ./cpp-capture, I'm getting this error

RealSense error calling rs_create_context(api_version:10907):
    api version mismatch

How do i get past it?

dorodnic, 2016-09-09:
This error occurs whenever the application (in this case cpp-capture) is not compiled using the same version of rs.h as the library that is being loaded (librealsense.so)
Please try to `make clean` and `make && sudo make install` again just to be sure. 99% of the times this is the problem. The remaining 1% is when something tricky is going on, like alternative path to librealsense.so in `LD_LIBRARY_PATH`
If this does not help, you can always take a closer look at what is going on inside `rs_create_context` (rs.cpp:50) either using GDB or by outputting `api_version` and `runtime_api_version` to console.

RickBlacker, 2016-09-09:
Hi, I just ran 

* make clean
* make && sudo make installl

Still getting the same issue. 
Is it possible that I need to remove the headers and library from /usr/local ???

RickBlacker, 2016-09-09:
Ok, tried manually removing library and header from /usr/local.  Then recompiled, no change. Still getting the same error.

dorodnic, 2016-09-13:
@RickBlacker, any news on this one?

ev-mp, 2016-09-13:
@RickBlacker, please also run 'sudo make uninstall' before 'make clean && make -j8 && sudo make install'. 
It could be that 'make install' fails to replace the older version  correctly, and that it requires an additional step to remove

BastianJaeger, 2016-09-26:
Same problem here. Did you solve it?

jacksonkr, 2016-10-22:
@BastianJaeger 

that folder corresponds to the package `ros-[distro]-librealsense`. I'm curious, if you remove that package via `apt-get` are you still able to keep everything working?

I'm running into this same issue with r200 on ubuntu 14.04 amd64 3.19.0-73-generic

adihayat, 2016-11-22:
Hi , getting the same issue as @jacksonkr ,
was this issue resolved ? 
`
/usr/local/bin/cpp-callback

RealSense error calling rs_create_context(api_version:11000):
    API version mismatch: librealsense.so was compiled with API version 1.11.0 but the application was compiled with 1.10.0! Make sure correct version of the library is installed (make install)
`

jacksonkr, 2016-11-24:
@adihayat so there is `librealsense` and `realsense_camera`.

`librealsense` is the core and `realsense_camera` interfaces the core driver with the camera. If they are different versions then you get the API version mismatch. This usually happens because you have a `librealsense.so` somewhere on your machine that is old. I'm not going to lie, getting it all to match up was a real pain in the ass so good luck.

closed 2017-10-26 by jlgarcia75