realsenseai/librealsense #283: Firmware problems using SR300 Development Kit

HugoCMU, 2016-09-09:
I purchased the SR300 Development kit online through this link:

http://click.intel.com/intelrrealsensetm-developer-kit-featuring-sr300.html

I ran into issues after following your install instructions (failed with `uvcvideo: Unknown video format` errors after running the `./scripts/patch-uvcvideo-16.04.simple.sh`).

I cannot load the relevant firmware on it because it is not recognized by the Intel Realsense DCM. I noticed that it comes up in the device manager as "Blaster Senz3D VF0810". Is this a different model entirely from the SR300?

The information online is very sparse, and if I can't even get it recognized by the DCM to upload firmware I am kind of screwed :( 

| Required Info  |   |
|---|---|
| Camera Model | SR300 (VF0810) | 
| Firmware Version |  Uknown  | 
| Operating System & Version | 16.04   |
| Kernel Version (Linux Only)  | 4.4  |
| Build System |  N.A. |

dorodnic, 2016-09-10:
Hi @HugoCMU 
As a starting point could you please share the results of `lsusb`, `ls /sys/class/video4linux` and `sudo dmesg | tail -n 50`. You should be able to see a device with vendor ID 8086 (Intel) and product ID 0aa5 (the SR300). 
Not all video formats required by the SR300 has been up-streamed, but `./scripts/patch-uvcvideo-16.04.simple.sh` should be able to address that. 
I have found a similar issue reported [here](https://software.intel.com/en-us/forums/realsense/topic/676330).
On Windows, do you see any new imaging devices in the device manager when connecting the camera?

HugoCMU, 2016-09-12:
I found another Windows 10 computer and I was able to get the camera recognized and the latest firmware v3.2.26.6137 installed. (Link: https://downloadcenter.intel.com/download/25044/Intel-RealSense-Depth-Camera-Manager). I tested the camera using the Windows SDK and both the depth and color camera are working. (Link: https://software.intel.com/en-us/intel-realsense-sdk/download)

Output of `lsusb` shows the Creative camera, as well as some USB3.0 hubs.

```
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 005: ID 8086:0aa5 Intel Corp. 
Bus 002 Device 004: ID 05e3:0616 Genesys Logic, Inc. hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 007: ID 041e:409b Creative Technology, Ltd 
Bus 001 Device 006: ID 05e3:0610 Genesys Logic, Inc. 4-port hub
Bus 001 Device 005: ID 093a:2532 Pixart Imaging, Inc. 
Bus 001 Device 003: ID 04ca:004b Lite-On Technology Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

Output of `ls /sys/class/video4linux` shows 2 video devices. Both of them are Intel Realsense cameras:

```
$ cat /sys/class/video4linux/video0/name
Intel(R) RealSense(TM) Camera S
$ cat /sys/class/video4linux/video1/name
Intel(R) RealSense(TM) Camera S
```

The relevant part of the `dmesg | tail` shows the camera connected, but there is still the unknown video format errors from before:

```
[317410.113859] usb 2-5: new SuperSpeed USB device number 4 using xhci_hcd
[317410.132677] usb 2-5: New USB device found, idVendor=05e3, idProduct=0616
[317410.132685] usb 2-5: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[317410.132689] usb 2-5: Product: USB3.0 Hub
[317410.132693] usb 2-5: Manufacturer: GenesysLogic
[317410.134635] hub 2-5:1.0: USB hub found
[317410.134924] hub 2-5:1.0: 2 ports detected
[317410.993717] usb 1-5.2: new full-speed USB device number 7 using xhci_hcd
[317411.170642] usb 1-5.2: config 1 has an invalid interface number: 3 but max is 2
[317411.170650] usb 1-5.2: config 1 has no interface number 2
[317411.240791] usb 1-5.2: New USB device found, idVendor=041e, idProduct=409b
[317411.240799] usb 1-5.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[317411.240804] usb 1-5.2: Product: BlasterX Senz3D VF0810
[317411.240808] usb 1-5.2: Manufacturer: Creative Technology Ltd
[317411.240811] usb 1-5.2: SerialNumber: 160603170234
[317411.385789] input: Creative Technology Ltd BlasterX Senz3D VF0810 as /devices/pci0000:00/0000:00:14.0/usb1/1-5/1-5.2/1-5.2:1.3/0003:041E:409B.0006/input/input25
[317411.442226] hid-generic 0003:041E:409B.0006: input,hiddev0,hidraw4: USB HID v1.11 Device [Creative Technology Ltd BlasterX Senz3D VF0810] on usb-0000:00:14.0-5.2/input3
[317412.569848] usb 2-5.1: new SuperSpeed USB device number 5 using xhci_hcd
[317412.586698] usb 2-5.1: New USB device found, idVendor=8086, idProduct=0aa5
[317412.586705] usb 2-5.1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[317412.586710] usb 2-5.1: Product: Intel(R) RealSense(TM) Camera SR300
[317412.586714] usb 2-5.1: Manufacturer: Intel(R) RealSense(TM) Camera SR300
[317412.588261] uvcvideo: Found UVC 1.10 device Intel(R) RealSense(TM) Camera SR300 (8086:0aa5)
[317412.590398] input: Intel(R) RealSense(TM) Camera S as /devices/pci0000:00/0000:00:14.0/usb2/2-5/2-5.1/2-5.1:1.0/input/input26
``` 
**`[317412.590982] uvcvideo: Unknown video format 435a4e49-b602-480f-978c-e4e88ae89b89`**
**`[317412.590989] uvcvideo: Unknown video format 52494150-8536-4841-b6bf-8fc6ffb083a8`** 
```
[317412.591052] uvcvideo: Found UVC 1.10 device Intel(R) RealSense(TM) Camera SR300 (8086:0aa5)
[317412.592797] uvcvideo: Unable to create debugfs 2-5 directory.
[317412.593021] input: Intel(R) RealSense(TM) Camera S as /devices/pci0000:00/0000:00:14.0/usb2/2-5/2-5.1/2-5.1:1.2/input/input27
```

HugoCMU, 2016-09-12:
Re-ran the `./scripts/patch-uvcvideo-16.04.simple.sh` script and it seems like the Product name on `dmesg` has changed, and is now SR300 as opposed to Creative BlasterX Senz3D.

```
[  172.199012] usb 2-5.1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  172.199017] usb 2-5.1: Product: Intel(R) RealSense(TM) Camera SR300
[  172.199021] usb 2-5.1: Manufacturer: Intel(R) RealSense(TM) Camera SR300
[  172.200844] uvcvideo: Found UVC 1.10 device Intel(R) RealSense(TM) Camera SR300 (8086:0aa5)
[  172.203172] input: Intel(R) RealSense(TM) Camera S as /devices/pci0000:00/0000:00:14.0/usb2/2-5/2-5.1/2-5.1:1.0/input/input26
[  172.203754] uvcvideo: Unknown video format 435a4e49-b602-480f-978c-e4e88ae89b89
[  172.203762] uvcvideo: Unknown video format 52494150-8536-4841-b6bf-8fc6ffb083a8
[  172.203827] uvcvideo: Found UVC 1.10 device Intel(R) RealSense(TM) Camera SR300 (8086:0aa5)
[  172.205653] uvcvideo: Unable to create debugfs 2-5 directory.
[  172.205935] input: Intel(R) RealSense(TM) Camera S as /devices/pci0000:00/0000:00:14.0/usb2/2-5/2-5.1/2-5.1:1.2/input/input27
```

ev-mp, 2016-09-13:
Hi @HugoCMU , please run "cpp-enumerate" demo and send the camera FW version: e.g.
"Device 0 - Intel RealSense SR300:
 Serial number: 548201002754
 **Firmware version: 3.16.3.0**
 USB Port ID: \\.\USB#VID_0424&PID_5..."

The unrecognized formats that you see in dmesg log are additinal SR300 formats used exclusively during production activities, and are not publicly available.  
We still need to understand why these descriptors haven't been removed correctly in your case, but other than dmesg warnings, it should not impact camera operation -  and you should be able to stream depth, infrared and color as advertised.

Please check and update.

HugoCMU, 2016-09-13:
The firmware version is as follows:
```
Device 0 - Intel RealSense SR300:
 Serial number: 611203002997
 Firmware version: 3.17.0.0
 USB Port ID: 2-1
```
The output of the ` SR300-live-test` script shows all of the tests passing, although there seems to be different results when run consecutively though...
```
All tests passed (39291 assertions in 42 test cases)
```

closed 2016-11-19 by dorodnic

Vibek, 2017-02-22:
Hello,
Our lab brought new Blaster-X sensz3D camera. After lots of struggle found the this camera not support linux. But, successfully access the camera using gstreamer.
Does any have any working solution to access the camera in Opencv or ROS.

dreamibor, 2018-01-22:
@Vibek Hi, have you figure out how to run the Blaster-X sensz3D camera on Linux?

Vibek, 2018-01-22:
Hey Zhaoyu,
I did able to run the the Blaster-X sensz3D camera on Linux. It works well
for me. Please have a look on my github page. Everything is written there.
How to run the camera on Linux. Note that I just made a small hack to run
the camera on Linux.  I am still working on a hack to access the both depth
and RGB images for computer vision problems. It is still not done.

https://github.com/Vibek/Blaster-X-Senz3D-camera-driver

dreamibor, 2018-01-22:
@Vibek Brilliant, thank you!