realsenseai/librealsense #284: Building using Makefile

MartenF, 2016-09-11:
| Operating System & Version | armbian |
| Kernel Version (Linux Only)  | 3.10.102-odroidxu4   |
| Build System |  Makefile | 

When trying to build the libary using make it's failing with the following error. Im using gcc version 4.9.2

> g++ src/image.cpp -std=c++11 -fPIC -pedantic -Ofast -Wno-missing-field-initializ                                                                             ers -Wno-switch -Wno-multichar -DRS_USE_V4L2_BACKEND `pkg-config --cflags --libs                                                                              libusb-1.0`  -mfpu=neon -mfloat-abi=hard -ftree-vectorize -c -o obj/image.o
src/image.cpp: In function ‘void rsimpl::unpack_rw10_from_rw8(rsimpl::byte* cons                                                                             t*, const byte*, int)’:
src/image.cpp:75:43: error: ISO C++ forbids declaration of ‘type name’ with no t                                                                             ype [-fpermissive]
         auto src = reinterpret_cast<const __m128i *>(s);
                                           ^
src/image.cpp:75:43: error: expected ‘>’ before ‘__m128i’
src/image.cpp:75:43: error: expected ‘(’ before ‘__m128i’
src/image.cpp:75:43: error: ‘__m128i’ was not declared in this scope
src/image.cpp:75:52: error: expected primary-expression before ‘>’ token
         auto src = reinterpret_cast<const __m128i *>(s);
                                                    ^
src/image.cpp:75:56: error: expected ‘)’ before ‘;’ token
         auto src = reinterpret_cast<const __m128i *>(s);
                                                        ^
src/image.cpp:76:37: error: ‘__m128i’ does not name a type
         auto dst = reinterpret_cast<__m128i *>(d[0]);
                                     ^
src/image.cpp:76:45: error: expected ‘>’ before ‘*’ token
         auto dst = reinterpret_cast<__m128i *>(d[0]);
                                             ^
src/image.cpp:76:45: error: expected ‘(’ before ‘*’ token
src/image.cpp:76:46: error: expected primary-expression before ‘>’ token
         auto dst = reinterpret_cast<__m128i *>(d[0]);
                                              ^
src/image.cpp:76:53: error: expected ‘)’ before ‘;’ token
         auto dst = reinterpret_cast<__m128i *>(d[0]);
                                                     ^
src/image.cpp:78:18: error: ‘xin’ was not declared in this scope
         __m128i* xin = (__m128i*)src;
                  ^
src/image.cpp:78:33: error: expected primary-expression before ‘)’ token
         __m128i* xin = (__m128i*)src;
                                 ^
src/image.cpp:79:18: error: ‘xout’ was not declared in this scope
         __m128i* xout = (__m128i*) dst;
                  ^
src/image.cpp:79:34: error: expected primary-expression before ‘)’ token
         __m128i* xout = (__m128i*) dst;
                                  ^
src/image.cpp:82:22: error: expected ‘;’ before ‘in1_16’
             __m128i  in1_16 = _mm_load_si128((__m128i*)(xin));
                      ^
src/image.cpp:83:22: error: expected ‘;’ before ‘in2_16’
             __m128i  in2_16 = _mm_load_si128((__m128i*)(xin + 1));
                      ^
src/image.cpp:84:22: error: expected ‘;’ before ‘out1_16’
             __m128i  out1_16 = _mm_srli_epi16(in1_16, 2);
                      ^
src/image.cpp:85:22: error: expected ‘;’ before ‘out2_16’
             __m128i  out2_16 = _mm_srli_epi16(in2_16, 2);
                      ^
src/image.cpp:86:22: error: expected ‘;’ before ‘out8’
             __m128i  out8 = _mm_packus_epi16(out1_16, out2_16);
                      ^
src/image.cpp:87:35: error: ‘out8’ was not declared in this scope
             _mm_store_si128(xout, out8);
                                   ^
src/image.cpp:87:39: error: ‘_mm_store_si128’ was not declared in this scope
             _mm_store_si128(xout, out8);
                                       ^
Makefile:106: recipe for target 'obj/image.o' failed
make: *** [obj/image.o] Error 1

dorodnic, 2016-09-11:
Hi @MartenF, 
Please note that librealsense doesn't officially support ARM based platforms. However, you are correct that this is a bug - the contents of `unpack_rw10_from_rw8` should be under `#ifdef __SSSE3__` and an alternative version should be provided. We will do our best of fix it soon or you are welcome to submit a pull-request if you wish.

mdhorn, 2016-09-13:
@dorodnic @MartenF  I'm submitting a PR to fix this.

mdhorn, 2016-09-13:
https://github.com/IntelRealSense/librealsense/pull/285

dorodnic, 2016-09-13:
Thank you, @mdhorn !

PR #285 (2016-09-13): Enable compilation for non-SSE platforms
Fixes: #284 

See commit 99053e8cec6764b59b78761ee3ce784768e4a4ec
closed 2016-09-13 by dorodnic