realsenseai/librealsense #287: Duplicate timestamp value for ZR300 camera color stream

reaganlo, 2016-09-13:
| Required Info  |   |
|---|---|
| Camera Model | ZR300 | 
| Firmware Version |  2.0.71.20 | 
| Operating System & Version | Ubuntu 14.04  |
| Kernel Version (Linux Only)  | 4.4.0-040400-generic  |
| Build System |  Makefile | 

For **ZR300** cameras, in **librealsense v1.9.7**, the **rs_get_frame_timestamp()** function always returns the same value for **color** streams.

dorodnic, 2016-09-14:
Hi @reaganlo 
Quick test did not reproduce this behavior (I just added `dev.get_frame_timestamp(rs::stream::color)` to cpp-capture and it is running at the proper frame-rate). Could you please provide more information? Also, please specify the exact hardware configuration you are using (first block of cpp-enumerate-devices)

reaganlo, 2016-09-15:
Hi @dorodnic, I was able to figure out the issue. 

It is because the **return type** for rs_get_frame_timestamp() was **int** in librealsense v0.9.2. But it has been changed to **double** in v1.9.7
So the data was getting truncated in our ROS package. I have modified the code in our ROS package to reflect this change. Hence I am closing this issue. Thank you.

closed 2016-09-15 by reaganlo